armbian/config #1: fix for softether

blindpet, 2015-12-26:
They must have changed something, regardless I've updated my fork to fix it

https://github.com/blindpet/MediaServerInstaller/commit/255bab1162212b4cb79a456ecf8eac6c133bb2ef

igorpecovnik, 2015-12-26:
Could be. This scripts need some update ... busy elsewhere :(

igorpecovnik, 2015-12-26:
I started on dev branch:
https://github.com/igorpecovnik/Debian-micro-home-server/commits/dev

blindpet, 2015-12-26:
Yea I know you're busy, trying to help where I can. Here's a fix for Varnish
```
wget -O - https://repo.varnish-cache.org/GPG-key.txt | apt-key add -
cat > /etc/apt/sources.list.d/varnish-cache.list<<EOF
deb-src https://repo.varnish-cache.org/debian/ jessie varnish-4.1
EOF
apt-get update
apt-get build-dep varnish -y
cd /tmp
apt-get source varnish -y
rm varnish_*.dsc
rm varnish_*.orig.tar.gz
rm varnish_*.diff.gz
cd varnish-4*
./configure --prefix=/usr
make -j$(ncpu)
make install
cp debian/varnish.init /etc/init.d/varnish
chmod +x /etc/init.d/varnish
cp debian/varnish.default /etc/default/varnish
update-rc.d varnish defaults
mkdir -p /etc/varnish
cp etc/example.vcl /etc/varnish/default.vcl
dd if=/dev/random of=/etc/varnish/secret count=1
service varnish start
```

closed 2018-10-21 by igorpecovnik