armbian/config #3: Error rpimonitor

Humanoide70, 2016-12-17:
Hello.
I have installed a mini home server, on a minibian distribution of 2016-03-12 and I have an error message:
Failed to start rpimonitor: Unit rpimonitor.service failed to load: Directory
functions.sh: line 283: /usr/share/rpimonitor/scripts/updatePackagesStatus.pl: No such file or directory
And the installation stops, and returns to the prompt.
Could you help me please?
Thanking you in advance.

J-Yves

igorpecovnik, 2016-12-17:
I assume you are using this on Raspberry pi? Is this minibian Debian Wheezy or Jessie?

Humanoide70, 2016-12-17:
Hi,
I use it on a minibian Jessie version of 2016-03-12. For systemd, where should it be?

igorpecovnik, 2016-12-17:
http://unix.stackexchange.com/questions/121654/convenient-way-to-check-if-system-is-using-systemd-or-sysvinit-in-bash

I can check later if there is something wrong with our script.

Humanoide70, 2016-12-17:
Maybe sysvinit when I check pid=1

Humanoide70, 2016-12-17:
What is the best os distribution?

igorpecovnik, 2016-12-18:
I don't use Raspberry PI and have no idea what is the best OS there. I'll try to fix this bug ASAP. In the mean time try to install from RPi monitor author Github: https://github.com/XavierBerger/RPi-Monitor-deb

CptnKickass, 2017-06-15:
+1 for this issue. On a RaspPi3 with systemd.

closed 2024-09-23 by igorpecovnik