armbian/config #8: /usr/bin/armbianmonitor: line 264: /sys/devices/system/cpu/cpu0/cpufreq/cpuinfo_cur_freq: No such file or directory

sanderjo, 2017-05-16:
`sudo armbian-config` -> `Monitor Simple CLI Monitoring` results in the repeated error message:

```
/usr/bin/armbianmonitor: line 264: /sys/devices/system/cpu/cpu0/cpufreq/cpuinfo_cur_freq: No such file or directory
```

Which is correct: there is no file `cpuinfo_cur_freq` in `/sys/devices/system/cpu/cpu0/`
```
sander@nanopineo2:~$ ll /sys/devices/system/cpu/cpu0/
total 0
drwxr-xr-x  6 root root    0 May 16 14:22 ./
drwxr-xr-x 10 root root    0 Jan  1  1970 ../
-rw-r--r--  1 root root 4096 May 16 14:20 cpu_capacity
drwxr-xr-x  2 root root    0 May 16 14:22 hotplug/
lrwxrwxrwx  1 root root    0 May 16 14:22 of_node -> ../../../../firmware/devicetree/base/cpus/cpu@0/
-rw-r--r--  1 root root 4096 May 16 14:22 online
drwxr-xr-x  2 root root    0 May 16 14:22 power/
drwxr-xr-x  3 root root    0 May 16 14:22 regs/
lrwxrwxrwx  1 root root    0 Jan  1  1970 subsystem -> ../../../../bus/cpu/
drwxr-xr-x  2 root root    0 May 16 14:22 topology/
-rw-r--r--  1 root root 4096 Jan  1  1970 uevent
```
FWIW:
```
sander@nanopineo2:~$ cpufreq-info
cpufrequtils 008: cpufreq-info (C) Dominik Brodowski 2004-2009
Report errors and bugs to cpufreq@vger.kernel.org, please.
analyzing CPU 0:
  no or unknown cpufreq driver is active on this CPU
  maximum transition latency: 4294.55 ms.
analyzing CPU 1:
  no or unknown cpufreq driver is active on this CPU
  maximum transition latency: 4294.55 ms.
analyzing CPU 2:
  no or unknown cpufreq driver is active on this CPU
  maximum transition latency: 4294.55 ms.
analyzing CPU 3:
  no or unknown cpufreq driver is active on this CPU
  maximum transition latency: 4294.55 ms.
sander@nanopineo2:~$ 
sander@nanopineo2:~$ 
sander@nanopineo2:~$ cpufreq-info -c 0
cpufrequtils 008: cpufreq-info (C) Dominik Brodowski 2004-2009
Report errors and bugs to cpufreq@vger.kernel.org, please.
analyzing CPU 0:
  no or unknown cpufreq driver is active on this CPU
  maximum transition latency: 4294.55 ms.
sander@nanopineo2:~$ 

```

I googled "no or unknown cpufreq driver is active on this CPU", but no solution.

Should I install software? Load a driver?

igorpecovnik, 2017-05-16:
Driver does not exits yet - you will have to make / port one.
Check:
https://www.armbian.com/nanopi-neo-2/
Known issues

closed 2017-05-16 by igorpecovnik

sanderjo, 2017-05-16:
Ah, I see

To avoid the error message if cpuinfo_cur_freq does not exist, would this be useful:

```
                if [ -f /sys/devices/system/cpu/cpu0/cpufreq/cpuinfo_cur_freq ]; then
                        CpuFreq=$(awk '{printf ("%0.0f",$1/1000); }' </sys/devices/system/cpu/cpu0/cpufreq/cpuinfo_cur_freq)
                else
                        CpuFreq=0
                fi
```

Works on my Nano Pi Neo2: "0MHz"