armbian/config #16: Patch, configs for update kernel 4.4.44 to 4.14.13 for BananaPi  and MuQSS patches

norrest, 2018-01-11:
Hi! 
My board is **Banana Pi+** with you image armbian but with patchet bfs kernel 
http://ck.kolivas.org/patches/bfs/
I install you image, download kernel from souce, use bfs patch, clean some modules from source, create *.deb`s and install
Now i look to install new kernel.
I download new kernel from kernel.org, unpack, than @make_old_config, patch with new MuQSS patches  
from http://ck.kolivas.org/patches/muqss/ and try to compile new deb pack`s. 
But i have many errors (
Can you help with compile new deb packs with MuQSS patch`s and some @old_configs? 
Commnd `fakeroot make -j5 deb-pkg LOCALVERSION=-mytest KDEB_PKGVERSION=1`  work for new kernel? Thank you

igorpecovnik, 2018-01-11:
Use our build script and try to add those patches to appropriate userpatches directory. If you want to compile natively on board you need all patches, proper compiler, ...

norrest, 2018-01-11:
Many thank you! Using your build  script, can I get the deb files? can I then install them?

igorpecovnik, 2018-01-11:
Yes. Deb files are default output.

norrest, 2018-01-13:
How to enable edit kernel config during using build script ?
patch need place to userpatches/   ? 
if i have .config how to provide it to new kernel? 
Thank you!!!!!

norrest, 2018-01-13:
[ o.k. ] * [u][c] 4.14-sched-MuQSS_162.patch
its ok?

norrest, 2018-01-13:
if i have .config how to provide it to new kernel?

norrest, 2018-01-13:
AR      fs/ubifs/ubifs.o
  AR      fs/ubifs/built-in.o
  AR      fs/built-in.o
[ error ] ERROR in function compile_kernel [ compilation.sh:341 ]
[ error ] Kernel was not built [ @host ]
[ o.k. ] Process terminated
  CC      fs/super.o
kernel/sched/MuQSS.c: In function ‘update_cpu_clock_tick’:
kernel/sched/MuQSS.c:2975:9: error: implicit declaration of function ‘get_irq_regs’; did you mean ‘get_mm_rss’? [-Werror=implicit-function-declaration]
  if (user_mode(get_irq_regs()))
         ^~~~~~~~~~~~
         get_mm_rss
kernel/sched/MuQSS.c:2975:24: error: invalid type argument of ‘->’ (have ‘int’)
  if (user_mode(get_irq_regs()))
                        ^~
cc1: some warnings being treated as errors
scripts/Makefile.build:314: recipe for target 'kernel/sched/MuQSS.o' failed
make[2]: *** [kernel/sched/MuQSS.o] Error 1
scripts/Makefile.build:573: recipe for target 'kernel/sched' failed
make[1]: *** [kernel/sched] Error 2
Makefile:1028: recipe for target 'kernel' failed
make: *** [kernel] Error 2
make: *** Waiting for unfinished jobs....

norrest, 2018-01-13:
can i put my old config to replace /root/build/config/kernel/linux-sunxi-next.config ?

igorpecovnik, 2018-01-13:
Here: userpatches/linux-sunxi-next.config

norrest, 2018-01-13:
No way to install MuQSS patch ? (

norrest, 2018-01-13:
Do I need to reinstall u-boot deb packages? I generated  linux-image-next-sunxi_5.37_armhf.deb and linux-headers-next-sunxi_5.37_armhf.deb
Can I use only them when I go?

igorpecovnik, 2018-01-13:
No need to replace uboot. Just remember to freeze your custom kernel to prevent beeing overwritten with our stock builds on apt update ...

norrest, 2018-01-13:
i create another linux-headers-next-sunxi_5.37_armhf.deb and linux-image-next-sunxi_5.37_armhf.deb for orange_pi_pc 
Created fine,  installed fine... But after reboot device is lost from lan network(

norrest, 2018-01-13:
[image]

norrest, 2018-01-13:
Selecting previously unselected package linux-u-boot-orangepipc-next.
dpkg: regarding linux-u-boot-next-orangepipc_5.37_armhf.deb containing linux-u-boot-orangepipc-next:
 linux-u-boot-orangepipc-dev conflicts with armbian-u-boot
  linux-u-boot-orangepipc-next provides armbian-u-boot and is to be installed.

dpkg: error processing archive linux-u-boot-next-orangepipc_5.37_armhf.deb (--install):
 conflicting packages - not installing linux-u-boot-orangepipc-next

norrest, 2018-01-13:
Sorry, for orange pi pc  this method work 
But for cubieboar A 20 - dont ((
**linux-dtb-next-sunxi_5.37_armhf.deb
linux-headers-next-sunxi_5.37_armhf.deb
linux-image-next-sunxi_5.37_armhf.deb
linux-u-boot-next-cubieboard2_5.37_armhf.deb**
install.. bo errors... but board dont load ( 
In boot folder some another structure of files for orange pi pc and cubieboar A 20/ banana pi
orange pi pc newest and after install new kernel debs - all links changed automatics
After install for cubieboar A 20/ banana pi - link must changed manually! 
I think problem in this!

norrest, 2018-01-13:
[image]
[image]

norrest, 2018-01-13:
earlier, after fixing the zImage reference, the installation of new kernels was successful, but now I can not update either myself or through your scripts

norrest, 2018-01-14:
linux-wheezy-root-next-cubieboard2 need forupdate?

norrest, 2018-01-14:
`
dpkg -i *.deb
Selecting previously unselected package linux-dtb-next-sunxi.
(Reading database ... 80302 files and directories currently installed.)
Preparing to unpack linux-dtb-next-sunxi_5.37_armhf.deb ...
Unpacking linux-dtb-next-sunxi (5.37) ...
Selecting previously unselected package linux-headers-next-sunxi.
Preparing to unpack linux-headers-next-sunxi_5.37_armhf.deb ...
Unpacking linux-headers-next-sunxi (5.37) ...
Selecting previously unselected package linux-image-next-sunxi.
Preparing to unpack linux-image-next-sunxi_5.37_armhf.deb ...
Unpacking linux-image-next-sunxi (5.37) ...
Selecting previously unselected package linux-u-boot-bananapi-next.
Preparing to unpack linux-u-boot-next-bananapi_5.37_armhf.deb ...
Unpacking linux-u-boot-bananapi-next (5.37) ...
Setting up linux-dtb-next-sunxi (5.37) ...
Setting up linux-headers-next-sunxi (5.37) ...
Compiling headers - please wait ...
Setting up linux-image-next-sunxi (5.37) ...
Setting up linux-u-boot-bananapi-next (5.37) ...
Updating u-boot on /dev/mmcblk0
root@volumio:~/kernel#

`

igorpecovnik, 2018-01-14:
Sorry, I lost track. Move summary here: https://forum.armbian.com/forum/4-development/ I am closing this issue since it's completely unrelated to armbian configuration script in a first place. I just realized that.

closed 2018-01-14 by igorpecovnik