armbian/config #18: Syncthing is configure to run as root when installed via armbian-config

PBRCW, 2018-03-09:
When syncthing is installed via arbiam-config the installation is made so that the service is run as root. This is discouraged by the syncthing team, see here:
[https://forum.syncthing.net/t/running-syncthing-as-root/6316](https://forum.syncthing.net/t/running-syncthing-as-root/6316)
So this should be default configured for the usage of an unpriviledged user.

ThomasKaiser, 2018-03-10:
Yes, [this](https://forum.syncthing.net/t/running-syncthing-as-root/6316/2) says it all.

Running as root is irresponsible and needs to be changed (checking `id`, using `${SUDO_USER}` for the whole stuff and asking which user account to choose in case the user directly logged in as root)

igorpecovnik, 2018-03-10:
Perhaps such implementation? https://github.com/armbian/config/blob/development/debian-config-submenu#L501-L521?

A part to create a new user/display error is missing.

ThomasKaiser, 2018-03-10:
I would check if `${SUDO_USER}` contains a valid user name. If that's the case this user will be used otherwise a dialog with available user accounts should be presented as you suggested. If there's none (only root) then I would stop and tell the user to create an unprivileged user account being member of `sudo` group first and then try again.

igorpecovnik, 2019-03-06:
Service doesn't run as root, closing.

closed 2019-03-06 by igorpecovnik