armbian/config #20: armbian-config -> install kernel headers fail

Aljumaili85, 2018-05-05:
Hi ,
armbian-config is not able to download kernel headers . I am using armbian for Allwinner A20 (Armbian ubuntu - legecy kernel 3.4.113-sun7i ) . any suggestion or even manual installation from terminal  ??

igorpecovnik, 2018-05-06:
Try using development branch of armbian-config which is accessible from the beta repository or check this page for manual installation: https://www.armbian.com/kernel/

Aljumaili85, 2018-05-06:
Thanks Igor, I manged to download it manually but  when I do make scripts nothing show as you explained on the bellowing post 
[image]

igorpecovnik, 2018-05-06:
Strange. Then open a support topic here: https://forum.armbian.com/forum/7-allwinner-a20/ Please provide all possible data most important: armbianmonitor -u

closed 2018-05-06 by igorpecovnik