armbian/config #23: error install armbian-stretch-desktop

carlosnewmusic, 2018-07-04:
Exception ignored in: <function WeakValueDictionary.__init__.<locals>.remove at 0xb6243420>
Traceback (most recent call last):
  File "/usr/lib/python3.5/weakref.py", line 117, in remove
TypeError: 'NoneType' object is not callable
Exception ignored in: <function WeakValueDictionary.__init__.<locals>.remove at 0xb6243420>
Traceback (most recent call last):
  File "/usr/lib/python3.5/weakref.py", line 117, in remove
TypeError: 'NoneType' object is not callable

igorpecovnik, 2018-07-04:
what about if you install it manually, apt install armbian-stretch-desktop ... and where did you try? Please provide all data (armbianmonitor -u) ... if you run Armbian

carlosnewmusic, 2018-07-05:
[image]
http://ix.io/1g8v

igorpecovnik, 2018-07-05:
I see you try to install a desktop on top of OMV, which changes system beyond recognition. For their needs. You are running many other services, which could compromise some system files. Hard to say. Our desktop is nothing special - it uses standard XFCE4 desktop + our theming.

I can't reproduce this error. I tried a few different setups. https://pastebin.com/54PMnFY6

closed 2018-07-05 by igorpecovnik

carlosnewmusic, 2018-07-05:
Error when configuring the xorg menu

igorpecovnik, 2018-07-05:
Absolutely no idea & unrelated to our desktop.