armbian/config #26: Cannot manually configure eth0 if no ipv4 address previsouly assigned

g-provost, 2018-07-19:
[select_default_interface() ](https://github.com/armbian/config/blob/b72ddca49214f277797cdf3ec85f108c08f96723/debian-config-functions-network#L559)function will skip any interface that doesn't have inet address. Therefore we cannot choose an interface without an existing IP address for manually configuring it... which doesn't make sense. 

`[[ -n $IPADDR && $IPADDR != "172.24.1.1" ]] &&	LIST+=( "${i[0]//[[:blank:]]/}" "${IPADDR}" )`

Why `-n $IPADDR` ? Any reason to ignore interface without address ?

igorpecovnik, 2018-07-19:
No reason. This whole part is not very well made and needs some refactoring.

g-provost, 2018-07-20:
Then can we remove this `-n $IPADDR` ? I will do a PR.

closed 2018-12-06 by g-provost