armbian/config #31: add legend for utility keys

Tungsteno74, 2018-08-27:
Hello,
As i'm not able to enable/disable hardware options (i've tried a lot of keys , but no one do the work), it's possible to add an help space similar to the one shown in the link?
I own a OrangePi Zero.

thanks

[https://www.jpaul.me/wp-content/uploads/2015/12/c240san-1.png](https://www.jpaul.me/wp-content/uploads/2015/12/c240san-1.png)

igorpecovnik, 2018-08-27:
I don't exactly understand what do you mean by that? You mean enabling and disabling I2C, UART, ... in armbian-config -> system -> hardware?

Tungsteno74, 2018-08-27:
Hello Igor, yes you have catched the problem. When i move with arrows on , for example, I2C1 mode and then press ENTER key the choosen option is not selected (no dot or asterisk between brakets) but, after i quit the hardware menu section the tool tell me to reboot the system to apply changes but no changes are made at reboot. I've tried several times and tested the corresponding pins on the board but nothing.

Could be that I use the wrong key to enable the mode?

igorpecovnik, 2018-08-27:
Aha, use space to select, then save and you need to reboot to make that work. Will add some comments there ...

Tungsteno74, 2018-08-27:
@igorpecovnik  Ho really, was so simple. 
This was the first thing that i tried  but for some reason has not been triggered correctly at the time. 

Perhaps it was due to the bugged interface that occasionally appears to me.... but, in any case, now works!
 
Thank you very much Igor.

closed 2018-08-27 by igorpecovnik