armbian/config #32: tv headend installation is failing

Marty56, 2018-09-23:
Maybe remove tv headend from the armbian config tool since it is not working.

igorpecovnik, 2018-09-23:
It works. I did install it a few days ago on a Bionic version. On the clean image of course. Where did you try to install it?

Marty56, 2018-09-23:
Thanks for the fast response and a potential fix.
But I had no luck. What I did: 

I tried apt update apt upgrade and the rerun armbian-config.
Selected tvheadend.

Some package installation dialog appeared but nothing else. I was back in the armbian-config menu.
My system is an orange pi plus 2E running armbian with the latest updates.

I tried to start tvheadend using 'sudo service tvheadend start' but receive 'Failed to start tvheadend.service: Unit tvheadend.service not found'

igorpecovnik, 2018-09-23:
> I tried apt update apt upgrade and the rerun armbian-config.

It doesn't go that fast. You need to wait for [next update](https://docs.armbian.com/Release_Changelog/) or grep sources from this repository - only there changes are present at this moment.

igorpecovnik, 2018-09-23:
> So I have to wait a couple of days until the next update is released.

Next stable release is not even planned. It can take weeks or months. You can switch to nightly repository (withing armbian-config) and it will show up in 24-48h ... if building of whole armbian suceeds. When switching to nightly build, you have to understand that those builds are not tested and remember, this service as is based. No warranty.

Marty56, 2018-09-23:
Ok. Then I would be very interested to grep sources from this repository.

would the instruction given in the ReadMe File under 
'Running this utility on 3rd party Debian based distributions' be the right approach?

igorpecovnik, 2018-09-23:
Buy a dinner https://www.armbian.com/donate and will push an update of armbian-config ASAP. More than this I can't help.

igorpecovnik, 2018-09-26:
You can apt update and upgrade to receive a fix from this issue.

Marty56, 2018-09-26:
Does not work. I see a package installer that stops at about 65 % and then it switches back to the 
armbian-config menu.

igorpecovnik, 2018-09-26:
It worked 3 days ago. Consider talking to Tvheadend makers and ask them what they fucked up. https://tvheadend.org/projects/tvheadend/wiki/AptRepository