armbian/config #33: Armbian-config hardware -> empty "Toggle hardware configuration" list

troyane, 2018-10-03:
I'm using `Armbian_5.59_Nanopiair_Debian_stretch_next_4.14.65.img` on [NanoPi Neo Air](http://wiki.friendlyarm.com/wiki/index.php/NanoPi_NEO_Air). 

System -> Hardware option looks like this:

[image]

Before showing this window with empty checkboxes it outputs this:
[image]

Same situation after switch to nightly automated builds.

igorpecovnik, 2018-10-08:
Could not recreate, closing. It must be related to bad SD card media or corrupted install.

Yadro, 2018-10-27:
`Armbian_5.65_Orangepizero_Debian_stretch_next_4.14.78.img` on Orange Pi Zero

The same issue with same log above.
I just installed OS, ran `apt update/upgrate` and set timezone and locals - `en-EN UTF-8` + `ru-RU UTF-8`

igorpecovnik, 2018-10-27:
Changing locales could have some impact on this. Have to check ...

Yadro, 2018-10-27:
Yep, I've reinstalled distr and with no changes of locals it's work.
Seems due to changes of locals we'll see empty list.

Yadro, 2018-10-27:
For Orange Pi Zero I see this list:
```
[ ] analog-codec        
[ ] cir                 
[ ] i2c0                
[ ] i2c1                
[ ] i2c2                
[ ] pps-gpio            
[ ] pwm                 
[ ] spdif-out           
[ ] spi-add-cs1         
[ ] spi-jedec-nor       
[ ] spi-spidev          
[ ] uart1               
[ ] uart2               
[ ] uart3               
[ ] usbhost0            
[ ] usbhost1            
[ ] usbhost1 
[*] usbhost2 
[*] usbhost3 
[ ] w1-gpio  
```
It can somebody help :)

igorpecovnik, 2018-10-27:
Good, we find the cause. Solution is that critical commands are called with LANG=C prefix and it should survive locales change. Like this one:  https://github.com/armbian/config/blob/master/debian-config-functions#L201

pwuertz, 2019-02-06:
I'm still seeing this exact error on a freshly downloaded / installed Armbian on a NanoPi Neo Air. Also tried `armbian-config` from github master, same error.

What is the current workaround? I tried setting `export LANG=C` before calling `armbian-config`, but it has no effect.

pwuertz, 2019-02-06:
I didn't change any locale settings by the way, this error is showing up on a vanilla armbian setup.

igorpecovnik, 2019-02-06:
I can reproduce this on a random occasion. Sometimes https://github.com/armbian/config/blob/master/debian-config-jobs#L1045 is executed properly, sometimes not. Don't know what is wrong :(

pwuertz, 2019-02-07:
Ok, I think I found something. On my affected system the `/boot/armbianEnv.txt` has beed corrupted, its contents got replaced by some PGP Private Key. After restoring `armbianEnv.txt` to something useful `armbian-config` works as expected.

I'm absolutely positive that I haven't touched that file. All I did was apt installing and apt update/upgrade once.

@igorpecovnik If you happen to experience this random occasion again, maybe you could have a look at `armbianEnv.txt` to check if this is indeed the same reason for this error. If so, there might be a rogue script somewhere in the first bootup or upgrade channel.

Gasman2014, 2019-02-12:
Exactly the same issue as @troyane here. The only thing that could have touched ` /boot/armbianEnv.txt` that I can think of is that I was investigating the other options in` armbian-config` and looked at the boot environment but did not write this back - apart from the `usbstoragequirks` entry it seemed to have acquired a ^A character but was otherwise empty. Copying a vanilla from another machine and retaining the `usbstoragequirks` entry now reboots and the system options selection now works.

```
verbosity=1
logo=disabled
console=both
disp_mode=1920x1080p60
overlay_prefix=sun8i-h3
overlays=usbhost2 usbhost3
usbstoragequirks .....
```

troyane, 2019-02-12:
@igorpecovnik @Gasman2014 I haven't seen this issue for a long time already. I've treated Igor's words `Could not recreate, closing. It must be related to bad SD card media or corrupted install.` as solution, so I've:
* got new microSD (SanDisk Ultra), 
* download newer available version of Armbian,
* burn Armbian using Etcher.

Gasman2014, 2019-02-12:
This is what I was getting

but is now OK since I have re-written my armbianEnv.txt with

verbosity=1
logo=disabled
console=both
disp_mode=1920x1080p60
overlay_prefix=sun8i-h3
overlays=uart1 uart2 usbhost2 usbhost3
usbstoragequirks=0x2537:0x1066:u,0x2537:0x1068:u

I hasten to add I have had NO other issues with my OPZero so I am not sure it is simply an SD card problem.

doublehp, 2019-03-25:
Consider this disucssion:

https://forum.armbian.com/topic/8819-where-to-report-a-bug-etcinitdarmhwinfo-corrupts-bootarmbianenvtxt-when-followed-by-powerfailure/?_fromLogin=1

It's not just ext4 or the SD that is broken; the issue occurs only when some init script alters armbianEnv.txt ; and the issue is 95% reproductible when you know how to: remove the line about USBquircks ... rerun the script that introduces it blindly, power cycle before sync.

One part of the problem is that this script may readd the line even if it's already in there.

I have fixed this by backing up the file regularly, and automatically restaure the last backup if it looks broken. For me this is now a fully automated service; backup files appear from time to time; but since I developped this service, I never had any failure.

An other part of the problem is how temp files are managed.

An other part of the problem is the lack of sync after some critical steps. Even if I know very well why sync should not be more frequent than 5mn on normal use, during the first 5mn after boot, there should be a sync every 10s. Especially after altering /boot ...

igorpecovnik, 2019-03-26:
>  power cycle before sync.

A sync can be added here and there, to make sure those configs are written properly.

>  there should be a sync every 10s

This will lead to another problem - extreme SD card lifespan shortening which will sooner or later produce similar problems - fs corruption. Bottom up. If you want to have file system that survive power cut, you need to alter few things and sacrifice performance/SD card endurance. Check / mount option in /etc/fstab

bonjour81, 2019-04-06:
Hi there,
I just switched from kernell 3.4.xxx to 4.11.xx via armbiam and my I2C seems not working anymore.
Anyway, I tried to check "toggle" hw options in armbian and I do have exact same display as in the 1st port from @troyane 

I'm not sure to have followed correctly the discussion here, but my armbianEnv.txt looks like this:
```
verbosity=7
console=both
machid=1029
bootm_boot_mode=sec
rootdev=UUID=f794393d-6f2d-48a6-a5a4-ba4f957a76c9
rootfstype=ext4
overlays=""
usbstoragequirks=0x2537:0x1066:u,0x2537:0x1068:u
```

Is there anything that I can do ?

I run an orange pi PC+ (with internal emmc, no sdcard)

doublehp, 2019-04-06:
bonjour81, I have seen many changes in the way I2C is handled between kernels 3 and 4. I have discussed about this at many places. From memory, you need to play with overlays in /boot/armbianEnv.txt . I had a lot of issues with SPI and 1W. I had a project which got stuck because one feature required a kernel 4 , and an other feature required kernel 3 for a driver which is not ported yet; project is stuck. Be carefull when playing with I2C, because a mistake can mess your eMMC; then you need to boot from SD to fix the eMMC. Also, armbian-config may not be up to date for the feature you need. And whatever you find in Google or a forum, assume it was written for kernel 3 and expect it to not work with kernel 4.

bonjour81, 2019-04-07:
Hi!
Thanks for informations !
I solved my I2C issue by replacing `overlays=""`  by `overlays=sun8i-h3-i2c0 to ` in armbianEnv.txt
My sensor is back :-) (even if I must say I'm not sure what I'm doing here...)

But the issue mentionned here is still present in armbian-config (no display in toggle HW section).
So far it's not blocking anymore for me, but if I can help, let me know how

doublehp, 2019-04-17:
"This will lead to another problem - extreme SD card lifespan shortening which will sooner or later produce similar problems - fs corruption."

Ony if you drop parts of the phrase you quoted: during the first 5mn after boot

ntfs1984, 2019-04-29:
I have the same issue with my cubietruck.
And it's definitely not SD issue, FS corruption or something like that.

[image]
[image]

dh1tw, 2019-06-01:
I had the same issue on several Orange Pi Zero H3 and H5. Thanks to @pwuertz comment, I can confirm that in all cases the file `/boot/armbianEnv.txt` was either overwritten by a PGP key or partially overwritten with some gibberish.. 
Replacing the file with an intact version for the particular platform fixed the issue. Now `armbian-config` works again properly.

ZupoLlask, 2019-12-28:
In my case, the culprit of the empty hardware section with `grep` errors was `overlay_prefix=` empty (in comparison with `overlay_prefix=sun50i-a64`) or totally missing at `/boot/armbianEnv.txt`.

I think armbian-config should check if that key is not missing or empty before running `grep`, and if needed it must try to recover that key and correct it at `/boot/armbianEnv.txt`.

As it is, it's not sufficiently robust.

igorpecovnik, 2019-12-28:
> it's not sufficiently robust.

I am aware of that, but since there is very little time and lots of issues it will remain this way. This is amateur development.  

Few help here and there changes a lot https://www.armbian.com/get-involved

ZupoLlask, 2019-12-28:
I'll see if I can create a decent PR for this... 😊

danielwilson654, 2019-12-30:
@ZupoLlask That would be great! This issue causes me all sorts of headaches.

danielwilson654, 2019-12-30:
I deploy armbian in the field where power cutoffs happen all to often. Would setting `/boot/armbianEnv.txt` to readonly resolve the issue for now?

GitTweaker, 2020-01-15:
My NanoPi NEO2 Black had the same issue.
Just added the line in /boot/armbianEnv.tx and the nit worked fine: 
overlay_prefix=sun50i-h5

chmondkind, 2020-01-24:
Had issues with my NanoPi Air. Resolved thanks to the information here:
sudo su
cd /boot/
mv armbianEnv.txt armbian.Env.txt.old
nano armbianEnv.txt
`overlay_prefix=sun8i-h3
overlays=usbhost2 usbhost3
rootdev=UUID=fc6d1d22-a58c-4f4b-89fd-b91068648fa3
rootfstype=ext4
usbstoragequirks=0x2537:0x1066:u,0x2537:0x1068:u`
Exit, save, reboot.

iav, 2020-05-13:
I found similar with nightly on ODroid-N2.
my `/boot/armbianEnv.txt` file was:
```
verbosity=1
logo=disabled
console=both
ethaddr=xx:xx:removed
rootdev=UUID=366ca76b-c168-4bea-97ef-b8d7efadcdb8
rootfstype=ext4
```
debug trace I add at theend of this message.
Error occures if `/boot/armbianEnv.txt` not contains `overlay_prefix` string at all or it have empty value.

It start to work when I add string `overlay_prefix=meson-`.

Seems there should be check for string presence with error diag — or kind of workaround for line 1220 of file `/usr/lib/armbian-config/jobs.sh`
```done < <(ls -1 ${OVERLAYDIR}/${overlay_prefix}*.dtbo | sed 's/^.*\('${overlay_prefix}'.*\)/\1/g' | sed 's/'${overlay_prefix}'-//g' | sed ...(truncated)
```
maybe I can try to fix this — if owner points me what behavior is correct for this case. 
```
+ case $1 in
+ CHANGES=false
+ true
++ cat /boot/armbianEnv.txt
++ grep overlay_prefix
++ sed s/overlay_prefix=//g
+ overlay_prefix=
+ TARGET_BRANCH=dev
+ exceptions dev
+ TARGET_FAMILY=meson64
+ UBOOT_BRANCH=dev
+ [[ dev == \d\e\f\a\u\l\t ]]
+ TARGET_BRANCH=-dev
+ [[ meson64 == pine64 ]]
+ [[ meson64 == sun*i ]]
+ MOTD=()
+ LINES=()
+ LIST_CONST=-3
+ j=0
+ DIALOG_CANCEL=1
+ DIALOG_ESC=255
+ read line
++ sed 's/^.*\(.*\)/\1/g'
++ ls -1 /boot/dtb/amlogic/overlay/meson-i2cA.dtbo /boot/dtb/amlogic/overlay/meson-i2cB.dtbo /boot/dtb/amlogic/overlay/meson-uartA.dtbo /boot/dtb/amlogic/overlay/meson-uartC.dtbo /boot/dtb/amlogic/overlay/meson-w1-gpio.dtbo /boot/dtb/amlogic/overlay/meson-w1AB-gpio.dtbo
++ sed s/.dtbo//g
++ sed s/-//g
+++ cat /boot/armbianEnv.txt
+++ grep overlays
+++ grep -w
Usage: grep [OPTION]... PATTERNS [FILE]...
Try 'grep --help' for more information.
++ [[ -n '' ]]
+ STATUS=
++ description ''
++ case $1 in
++ echo ''
+ DESC=
+ MOTD+=("$line" "$DESC" "$STATUS")
+ LINES[ $j ]=
+ ((  j++  ))
+ read line
+++ cat /boot/armbianEnv.txt
+++ grep overlays
+++ grep -w
Usage: grep [OPTION]... PATTERNS [FILE]...
Try 'grep --help' for more information.
++ [[ -n '' ]]
+ STATUS=
++ description ''
++ case $1 in
++ echo ''
+ DESC=
+ MOTD+=("$line" "$DESC" "$STATUS")
+ LINES[ $j ]=
+ ((  j++  ))
+ read line
+++ cat /boot/armbianEnv.txt
+++ grep overlays
+++ grep -w
Usage: grep [OPTION]... PATTERNS [FILE]...
Try 'grep --help' for more information.
++ [[ -n '' ]]
+ STATUS=
++ description ''
++ case $1 in
++ echo ''
+ DESC=
+ MOTD+=("$line" "$DESC" "$STATUS")
+ LINES[ $j ]=
+ ((  j++  ))
+ read line
+++ cat /boot/armbianEnv.txt
+++ grep overlays
+++ grep -w
Usage: grep [OPTION]... PATTERNS [FILE]...
Try 'grep --help' for more information.
++ [[ -n '' ]]
+ STATUS=
++ description ''
++ case $1 in
++ echo ''
+ DESC=
+ MOTD+=("$line" "$DESC" "$STATUS")
+ LINES[ $j ]=
+ ((  j++  ))
+ read line
+++ cat /boot/armbianEnv.txt
+++ grep overlays
+++ grep -w
Usage: grep [OPTION]... PATTERNS [FILE]...
Try 'grep --help' for more information.
++ [[ -n '' ]]
+ STATUS=
++ description ''
++ case $1 in
++ echo ''
+ DESC=
+ MOTD+=("$line" "$DESC" "$STATUS")
+ LINES[ $j ]=
+ ((  j++  ))
+ read line
+++ cat /boot/armbianEnv.txt
+++ grep overlays
+++ grep -w
Usage: grep [OPTION]... PATTERNS [FILE]...
Try 'grep --help' for more information.
++ [[ -n '' ]]
+ STATUS=
++ description ''
++ case $1 in
++ echo ''
+ DESC=
+ MOTD+=("$line" "$DESC" "$STATUS")
+ LINES[ $j ]=
+ ((  j++  ))
+ read line
+ exec
++ dialog --backtitle 'Configuration utility, Armbian 20.05.0-trunk.132 nightly, 192.168.1.220' --colors --title 'Toggle hardware configuration' --clear --cancel-label Back --ok-label Save --checklist '\nUse \Z1<space>\Z0 to toggle functions and save them. Exit when you are done.\n ' 0 0 0 '' '' '' '' '' '' '' '' '' '' '' '' '' '' '' '' '' ''
```

gigobait, 2022-12-24:
Has anyone fixed this? Every single image for the Renegade i burned to SD has the issue.

EvilOlaf, 2022-12-24:
Probably not and probably won't anymore since all efforts going into the overhaul of that tool as [announced in forums](https://forum.armbian.com/topic/25132-armbian-configurator-refactor-project/#comment-155591).

gigobait, 2022-12-25:
fix for Renegade

create
/boot/armbianEnv.txt                                                                                                                                                                                                                             

verbosity=1
bootlogo=true
overlay_prefix=rockchip
rootdev=UUID=e39e2e68-b810-41f9-8fd3-75e1ba55eb3c (get this by using blkid)
rootfstype=ext4
usbstoragequirks=0x2537:0x1066:u,0x2537:0x1068:u

save, reboot, the menu will start working. 
probably same concept for different boards, use the same family of chip