armbian/config #35: Behind http proxy

rvalle, 2018-10-15:
armbian-config should work behing an internet proxy. i.e. when http_proxy is defined, etc.

I am installing one board at work, and I could not workout why armbian-config is not working.

The line:

```
echo -e "GET http://github.com HTTP/1.0\n\n" | nc github.com 80 > /dev/null 2>&1
```
It is failing to detect that the firewall is cutting access to port 80, and armbian-config just hungs without returning any error.

epsilonrt, 2019-03-06:
Hello,
I have the same problem. My students can not use armbian-config in our lab rooms because it blocks on this line.
In an industrial environment (or military, or spacial ...), it seems to me that we should take into account its use in a context without internet connection, for example, we could use cached data when an internet connection is available (when installing the package at a minimum), then updated each time you launch armbian-config with internet access. So we could use it without internet connection.
Thanks

epsilonrt, 2019-03-06:
Immediately, I corrected the line by replacing it with

`wget -q --spider http: // github.com`

it works behind our proxy.

closed 2019-03-06 by igorpecovnik