armbian/config #37: Softy: OMV install fails, dependency issue

trohnjavolta, 2018-11-21:
HW: Odroid HC2
Armbian Version: 5.58, debian stretch

Tried to install omv 4 via armbian-config today and got this error:

```
Unpacking openmediavault-omvextrasorg (4.1.13) ...
dpkg: dependency problems prevent configuration of openmediavault-omvextrasorg:
 openmediavault-omvextrasorg depends on openmediavault (>= 4.0.4); however:
  Package openmediavault is not installed.

dpkg: error processing package openmediavault-omvextrasorg (--install):
 dependency problems - leaving unconfigured
Errors were encountered while processing:
 openmediavault-omvextrasorg
```
Is there a hotfix I could apply? Like swap install lines, so openmediavault gets installed before openmediavault-omvextrasorg?

trohnjavolta, 2018-11-21:
Now if I execute apt upgrade I get this:

```
Reading package lists... Done
Building dependency tree
Reading state information... Done
The following packages will be REMOVED:
  openmediavault-omvextrasorg
0 upgraded, 0 newly installed, 1 to remove and 0 not upgraded.
1 not fully installed or removed.
After this operation, 477 kB disk space will be freed.
Do you want to continue? [Y/n] y
(Reading database ... 48588 files and directories currently installed.)
Removing openmediavault-omvextrasorg (4.1.13) ...
/var/lib/dpkg/info/openmediavault-omvextrasorg.postrm: 22: .: Can't open /usr/share/openmediavault/scripts/helper-functions
dpkg: error processing package openmediavault-omvextrasorg (--remove):
 subprocess installed post-removal script returned error exit status 2
Errors were encountered while processing:
 openmediavault-omvextrasorg
E: Sub-process /usr/bin/dpkg returned an error code (1)
```
I checked, there's no /usr/share/openmediavault/scripts directory....
How can I fix this?

igorpecovnik, 2018-11-22:
OMV was working fine. We haven't changed anything which leads to believe there is an upstream problem. Are you installing this on a clean image?

trohnjavolta, 2018-11-22:
No, I have to reinstall omv. I believe a script I used messed with php files. It also installed a manually compiled php 7.2 version for stretch. Now I moved all apps I installed with the script to docker containers. From omv forum I got the advice to start over:
https://forum.openmediavault.org/index.php/Thread/24856-PHP-Fatal-Error-Uncaught-Error/?postID=188601#post188601

I removed plugins within omv and then purged it, also php 7.2. So far no errors came up.
I chose this way because I wanted to have the armbian improvements again for omv.

igorpecovnik, 2018-11-22:
Yes, those scripts works only on a clean Debian. Let's close the issue then.

closed 2018-11-22 by igorpecovnik

trohnjavolta, 2018-11-22:
Ok..well, how can I make it clean again? xD
I don't plan on reinstalling armbian and starting from zero, that defeats the purpose of me doing daily, weekly and monthly rsync backups.
I realise this is not the place to ask, but maybe someone could help me over at armbian forum:
https://forum.armbian.com/topic/8779-odroid-hc2-php-fatal-error-in-omv-how-to-reinstall-omv/ 
Posted there couple days ago but got impatient in the meantime.

igorpecovnik, 2018-11-22:
This is Linux and OMV is something we don't support here.