armbian/config #38: Syncthing-inotify to be removed?

catkfr, 2018-12-01:
Hi,

I have installed syncthing using softy on an odroid HC2. I am having syncthing-inotify configuration issues, but most importantly, I believe softy should remove syncthing-inotify from it's installation since it is officially depreciated:
> As of version v0.14.40, the functionality of syncthing-inotify is integrated in Syncthing and can be enabled in the advanced configuration menu for individual folders.

Just to be complete: this is what I tried:
I started investigating this issue:
`Dec 01 15:09:43 odroid systemd[1]: /etc/systemd/system/syncthing-inotify@user.service:14: Unknown lvalue 'Requires' in section 'Install'`

I tried to "fix" the syncthing-inotify configuration file by looking at the one in [github](https://github.com/syncthing/syncthing-inotify/blob/master/etc/linux-systemd/system/syncthing-inotify%40.service). Note that the Requires section is in Unit, not Install.

I was looking to uninstall syncthing-inotify and discovered it was not installed via the package manager. For those looking into this, you need to know that softy is just [this bash script](https://github.com/armbian/config/blob/master/debian-software)

igorpecovnik, 2019-03-06:
Fixed with https://github.com/armbian/config/commit/6a2567ddd0264b24d0b3b875c9c4397d7ae71ce4

closed 2019-03-06 by igorpecovnik

catkfr, 2019-03-07:
Thanks for the update!