armbian/config #43: install hotspot on raspberry 3b+

dragonlost, 2019-02-15:
Hello 

I install armbian-config on my raspberry 3b +.
I would like to use the feature to create an access point.

But unfortunately the script crashes before the end because of dnsmasq.

I am on ubuntu mate 16.04.5.

error message : `Job for dnsmasq.service failed because the controle process exited with error code. See "systemctl status dnsmasq.service" ans "journalctl -xe" for details`

systemctl status dnsmasq.service : 
```
dnsmasq : failed to create listening socket for 172.24.1.1 : Cannot assign requested address
Failed to start dnsmasq - A lightweight DHCP and caching DNS server
```

on journalctl -xe : 
`wpa_supplicant[689]: wlan0: Failed to initiate sched scan`

igorpecovnik, 2019-02-15:
Can you provide content of /etc/dnsmasq.conf ?

dragonlost, 2019-02-15:
interface=wlan0
listen-address=172.24.1.1
blind-interfaces
server=8.8.8.8
domain-needed
bogus-priv
dhcp-range=172.24.1.50,172.24.1.150,12h

igorpecovnik, 2019-02-16:
Probably wlan0 device lack its interface and that one is not created due lack of ifupdown packages or some other diff in the network configuration. Try to apt install ifupdown if that helps. In case not, this tool need some additional development to be able to run well outside its natural environment (Armbian).

dragonlost, 2019-02-25:
I have just returned from vacation. I checked ifupdown is well installed.

closed 2024-09-23 by igorpecovnik