armbian/config #46: Howto update Syncthing installed w. armbian-config ?

kreisklasse, 2019-03-06:
I installed Syncthing via the cli with _armbian-config_.
It first installed the release candidate (_v1.1.0-rc.1_)
In Syncthing-Settings i changed it to _"stable"_ only.

Now a new stable version showed up (_1.1.1_), but i can not update it.
Syncthing shows on the web-page:  
`upgrading: open /usr/bin/syncthing937255095: read-only file system`

**How can i update (manually/automatically) to the new Syncthing version?**

```
BOARD=odroidxu4
BOARD_NAME="Odroid XU4"
BOARDFAMILY=odroidxu4
VERSION=5.75
LINUXFAMILY=odroidxu4
BRANCH=next
ARCH=arm
IMAGE_TYPE=stable
BOARD_TYPE=conf
INITRD_ARCH=arm
KERNEL_IMAGE_TYPE=zImage

### boot environment:

Sun Feb 10 03:40:00 UTC 2019 | Odroid XU4 | 5.75 | armhf | armv7l | 4.14.94-odroidxu4
```

kreisklasse, 2019-03-06:
I added the Syncthing-Repo ( https://apt.syncthing.net/ ) but it still shows me no upgradable syncthing.

So how to update Syncthing in /usr/bin ?

igorpecovnik, 2019-03-06:
I guess at the time installer was made there were no packages. Installation is made this way: https://github.com/armbian/config/blob/master/debian-software#L815-L855
... which pools latest version from the Github. It need to be checked if their stock method still need any adjustments.

While it should be possible to upgrade since binaries are placed to the same directory ...

> read-only file system

Perhaps there is something wrong with your flash media or you are running some read only rootfs extensions?

igorpecovnik, 2019-03-06:
Fixed for future, closing.

closed 2019-03-06 by igorpecovnik