armbian/config #51: Armbian_5.83_Orangepizero_Ubuntu Setting zh_CN.UTF-8 UTF-8    failure

alog148, 2019-05-17:
I failed to set the language in version 5.83, which prevented me from entering Chinese. I did not see this problem by going back to 5.75 via the [mirror source](https://mirrors.tuna.tsinghua.edu.cn/armbian-releases/orangepizero/archive/)

igorpecovnik, 2019-05-17:
Check manual way of changing language first. Does it work? 

It has to be investigated ... We didn't change anything in this regard. It could be a bug in locales/language settings which is outside Armbian domain. Check also for solution and fix upstream at Debian/Ubuntu forums, their bug lists.

TiJey, 2019-05-20:
Good day! I faced the same problem on Orange Pi 3. You cannot change the language to ru_RU.UTF-8, not through armbin-config, not through dpkg-reconfigure locale, not through manual file editing.

igorpecovnik, 2019-05-20:
>  not through dpkg-reconfigure locale, not through manual file editing.

OK. In this case we can only wait that is fixed upstream. We don't have resources to fix bugs after 1000+ people''s work.

closed 2020-12-17 by igorpecovnik