armbian/config #52: Can't install NCP via armbian-config

Poincare1, 2019-05-18:
Going to Software > Softy > NCP > Install, outputs dialog box with "Loading system info..." then goes back to Softy selection screen.  NextCloudPersonal is not installed.  There are no errors onscreen. NCP has not been installed because no actions were taken.  

Other Softy(ware) installed include Samba and OMV.  

Is there a means for installed NCP?

igorpecovnik, 2019-05-18:
We only call their official installer https://github.com/armbian/config/blob/master/debian-software#L527

Check if their way of installing works:
https://github.com/nextcloud/nextcloudpi

Perhaps a method has to be changed?

Poincare1, 2019-05-18:
OK. Thanks for reply.  I looked at official installer.  What do I need to do?  
There is a bash .sh file that needs to be run before armbian-config?  It seems to indicate they support Odroid, which I am using and Debian, which I believe Armbian is based off of. 

I'm not sure what 'method' to be changed you are referring to.  All I'm trying to do is install NCP through Armbian-config.

igorpecovnik, 2019-05-18:
> We don't maintain this software 

We don't maintain NCP. You have to ask there why it doesn't work.

igorpecovnik, 2019-05-19:
Tested on ARM64 and ARMHF (Odroid XU4) Debian Stretch, works on both which means something must be wrong with your install. Perhaps a network issue. Closing.

closed 2019-05-19 by igorpecovnik