armbian/config #53: Cannot set static IP on unconfigured network interface

g-provost, 2019-05-27:
Regression.

Again we cannot set static IP on an unconfigured network interface because of the following test _-n $IPADDR_

https://github.com/armbian/config/blob/1fd13c254e45d0d2dea10c4fc4564aa8529ba56b/debian-config-functions-network#L566

Is there other use case where this test _-n $IPADDR_ is required ? Because it was removed previously but is back again via this commit : 8c79473df3f9cd1b78d0d5f0f5b166ab2733f032

igorpecovnik, 2019-05-29:
This function was never implemented so it could not broke down.

g-provost, 2019-05-29:
No it wasn’t an anomaly, I fixed it already last time through the following PR https://github.com/armbian/config/pull/27

g-provost, 2019-06-06:
For the time being can we remove the test **-n $IPADDR** ?

igorpecovnik, 2019-06-06:
That is not enough change - it will not help. I already started with refactoring to solve the problem but can't find the time to finish. I hope today.

g-provost, 2019-07-08:
Any ETA ? :P Sorry for pushing, kinda need this feature for our next upcoming batch.

igorpecovnik, 2019-07-08:
It's done in "development" branch but I had no time to test and debug.

g-provost, 2019-07-08:
Ok I don't know why I didn't see it the other day. I will test on my side and let you know.

g-provost, 2019-07-09:
I did some test. First I can say that the now the config tool choose the right interface (eth0 in my case). But few comments :
* Took me some trial to realize that the netmask field only accept CIDR notation because nmcli tool only accept CIDR notation. So I will add a function to convert netmask into cidr when necessary because otherwise the user no clue what he did wrong.
* It seems now everything mainly rely on NetworkManager, therefore old _ip_editor_ could be removed since it will never get called.
* When systemd-networkd.service is running then it's _systemd_ip_editor_ that is going to be called instead of _nm_ip_editor_. There is bug somewhere _systemd_ip_editor_ that make it exit as soon as it's called.

g-provost, 2019-07-09:
> * When systemd-networkd.service is running then it's systemd_ip_editor that is going to be called instead of nm_ip_editor. There is bug somewhere systemd_ip_editor that make it exit as soon as it's called.

Sorry not a bug, but I'm not sure that it's correct to expect that for example /etc/systemd/network/10-eth0.network exist.

https://github.com/armbian/config/blob/c353b94ba434f6c9e8a4b3405856a677e3b5cfe0/debian-config-functions-network#L378-L380

g-provost, 2019-07-10:
@igorpecovnik Could help me to understand why the following changes doesn't work. Same code works perfectly when I test it in a dedicated bash script, but once I add it to _nm_ip_editor_ in debian-config-functions-network the call to function _netmask_to_cidr_ function return me `(standard_in) 1: syntax error`. It seems the error comes from *bc* call. Maybe it's because of this `exec 3>&1`

https://github.com/helios-4/config/commit/3f0af47efa89861d77f627caa85fb270d1d91576

I run out of ideas :/

g-provost, 2019-07-11:
> I run out of ideas :/

Ok the CIDR conversion works now. Damn I spent ages looking in wrong direction. It was simply the IFS env var that wasn't setup correctly for new function _netmask_to_cidr_

I tested most of the use case and it works well. **We can merge Dev branch into Master.**

closed 2019-07-12 by g-provost