armbian/config #55: OMV doesn't work on Debian Buster

igorpecovnik, 2019-07-06:
@ThomasKaiser Shell we just remove it https://github.com/armbian/config/blob/4254f29201d4be9a56427a64ed130ea381f95a71/debian-software#L147 for Buster? For now.

ThomasKaiser, 2019-07-07:
It worked last week. Then some pushes to OMV repo resulted even in bricked updates from OMV 5.0.2 to 5.0.3. And I suppose it will work in the near future again.

What about adding for buster a warning/dialog reading 'Please be aware that OMV5 is still in beta state, might be functionally broken or not working at all. Report any issues you run into at forum.openmediavault.org'?

g-provost, 2019-07-08:
@ThomasKaiser Should we then point the script to install OMV 4.x (Arrakis) for Buster until OMV5 is not beta anymore ? I need to test if OMV4 install properly on Buster.

ThomasKaiser, 2019-07-08:
OMV4 is meant to run on Debian Stretch while OMV5 is meant to run on Buster. No idea whether OMV4 is able to run on Debian 10 but it's pretty easy to test out by adjusting the case construct in armbian-config's installation routine.

g-provost, 2019-07-08:
Ok I will test and let you know guys.

ThomasKaiser, 2019-07-08:
Some RPi 4 users report OMV4 [seems to work fine](https://forum.openmediavault.org/index.php/Thread/27710-Raspberry-pi-4-announced-better-than-3/?postID=207423#post207423) on Buster.

In the past those 'seems to work fine' reports when testing freshly released images turned out to several severe issues users simply overlooked or even forgot to report. So I'm not that convinced this will work. Also once OMV5 is ready a last OMV4 update will then contain an `omv-release-upgrade` command which tries to upgrade both OMV and Debian version at the same time and will most probably fail on OMV4/Buster.

TL;DR: If you want to run OMV4 remain on Stretch.

g-provost, 2019-07-15:
OMV5 install failed on Buster because it depends on **monit** package that doesn't exist in Buster.

Ok as you mentioned in issue #58 I agree this is a issue to be resolved by OMV team. However we should keep this issue open as a reminder to check the status once in a while.

ThomasKaiser, 2019-07-15:
> OMV5 install failed on Buster because it depends on monit package that doesn't exist in Buster.

See https://github.com/openmediavault/openmediavault/issues/410 for details.

g-provost, 2019-07-16:
Thanks for linking, I didn't look at OMV github issues.

I try with the monit dpkg from sid repo and install go through. Some minor errors here and there during the install but nothing nasty.

However openmediavault doesn't work. I need some digging to understand what service(s) not coming up, but currently I have nothing binding port 80. 
Also the install flush NetworkManager config, therefore in my case since I'm using a manually configured address, at next reboot, I ended up with eth interface unconfigured.

closed 2019-07-30 by g-provost