armbian/config #59: Plex install script shouldnt use dev2day.de repo anymore

g-provost, 2019-07-15:
According to dev2day.de maintainer, we shouldn't use anymore his repo.
This the message that appear while still installing form his repo.

> ##### ATTENTION #####
> 
> THIS IS THE LAST UPDATE OF THE dev2day.de PACKAGE!!! UPDATE TO THE OFFICIAL PACKAGE NOW!!!
> 
> Finally, official ARMv7 and ARMv8 Plex packages are available:
> https://www.plex.tv/media-server-downloads/
> 
> The migration is simple and your Plex library will be preserved.
> 
> First, remove the dev2day.de package:
> sudo apt-get remove plexmediaserver-installer
> 
> Then, remove the dev2day.de repo, e.g. with:
> sudo rm /etc/apt/sources.list.d/pms.list
> 
> Now, download the appropriate package from https://www.plex.tv/media-server-downloads/ and put it on your device, e.g.:
> wget https://downloads.plex.tv/plex-media-server-new/1.15.3.876-ad6e39743/debian/plexmediaserver_1.15.3.876-ad6e39743_armhf.deb
> 
> Finally, install the package, e.g.:
> sudo dpkg -i plexmediaserver_1.15.3.876-ad6e39743_armhf.deb
> 
> Also, the official Plex repository is now available:
> https://support.plex.tv/articles/235974187-enable-repository-updating-for-supported-linux-server-distributions/
> 
> The following thread has more information: https://forums.plex.tv/t/read-me-first-about-server-armv7-and-armv8-ubuntu-debian/226567/
> 
> Post your questions here: https://forums.plex.tv/tags/server-linux-arm
> 
> ####################
>

g-provost, 2019-07-15:
Latest official package armhf package from Plex website is : [plexmediaserver_1.16.1.1291-158e5b199_armhf.deb](https://downloads.plex.tv/plex-media-server-new/1.16.1.1291-158e5b199/debian/plexmediaserver_1.16.1.1291-158e5b199_armhf.deb)

I'm going to test and report here.

g-provost, 2019-07-15:
Ok. I installed and tested Plex by adding the official repo. I will update debian-software script to use official repo. I will need someone to test on arm64 though before we merge to master.

g-provost, 2019-07-15:
Ok I did the [modification](https://github.com/armbian/config/commit/26eaff6934aeec3ed981c1ff5f71a59b4b408b55) in the dev branch and tested on armhf. Works perfectly.
Can someone test on arm64 ?

g-provost, 2019-07-17:
Ok tested on arm64 (Rockpro64) and install and app work.

igorpecovnik, 2019-07-17:
tested also on amd64