armbian/config #60: NextcloudPi installer fails from Config -> 3rd Party Software

sunjam, 2019-07-15:
Using the basic curl installation method embedded in Armbian menu.
`# curl -sSL https://raw.githubusercontent.com/nextcloud/nextcloudpi/master/install.sh | bash`

No errors displayed when installing from Armbian, but when running curl script directly I see:

Hit:5 https://apt.armbian.com stretch InRelease
Reading package lists... Done
Reading package lists... Done
Building dependency tree       
Reading state information... Done
apt-utils is already the newest version (1.4.9).
cron is already the newest version (3.0pl1-128+deb9u1).
curl is already the newest version (7.52.1-5+deb9u9).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Reading package lists... Done
Building dependency tree       
Reading state information... Done
apache2 is already the newest version (2.4.25-3+deb9u7).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package php7.2
E: Couldn't find any package by glob 'php7.2'
E: Couldn't find any package by regex 'php7.2'
E: Unable to locate package php7.2-curl
E: Couldn't find any package by glob 'php7.2-curl'
E: Couldn't find any package by regex 'php7.2-curl'
E: Unable to locate package php7.2-gd
E: Couldn't find any package by glob 'php7.2-gd'
E: Couldn't find any package by regex 'php7.2-gd'
E: Unable to locate package php7.2-fpm
E: Couldn't find any package by glob 'php7.2-fpm'
E: Couldn't find any package by regex 'php7.2-fpm'
E: Unable to locate package php7.2-cli
E: Couldn't find any package by glob 'php7.2-cli'
E: Couldn't find any package by regex 'php7.2-cli'
E: Unable to locate package php7.2-opcache
E: Couldn't find any package by glob 'php7.2-opcache'
E: Couldn't find any package by regex 'php7.2-opcache'
E: Unable to locate package php7.2-mbstring
E: Couldn't find any package by glob 'php7.2-mbstring'
E: Couldn't find any package by regex 'php7.2-mbstring'
E: Unable to locate package php7.2-xml
E: Couldn't find any package by glob 'php7.2-xml'
E: Couldn't find any package by regex 'php7.2-xml'
E: Unable to locate package php7.2-zip
E: Couldn't find any package by glob 'php7.2-zip'
E: Couldn't find any package by regex 'php7.2-zip'
E: Unable to locate package php7.2-fileinfo
E: Couldn't find any package by glob 'php7.2-fileinfo'
E: Couldn't find any package by regex 'php7.2-fileinfo'
E: Unable to locate package php7.2-ldap
E: Couldn't find any package by glob 'php7.2-ldap'
E: Couldn't find any package by regex 'php7.2-ldap'
E: Unable to locate package php7.2-intl
E: Couldn't find any package by glob 'php7.2-intl'
E: Couldn't find any package by regex 'php7.2-intl'
E: Unable to locate package php7.2-bz2
E: Couldn't find any package by glob 'php7.2-bz2'
E: Couldn't find any package by regex 'php7.2-bz2'
E: Unable to locate package php7.2-json
E: Couldn't find any package by glob 'php7.2-json'
E: Couldn't find any package by regex 'php7.2-json'

igorpecovnik, 2019-07-17:
Please report the problem to NextcloudPi staff. We don't maintain their software. PHP7.2 is not default in (Armbian) Debian Stretch.

closed 2019-07-17 by igorpecovnik

nachoparker, 2019-07-18:
thanks, this happened because we are transitioning to Buster. Merging shortly

nachoparker, 2019-07-18:
merged

jimgregory, 2019-07-21:
Not working for me using an Orange Pi PC+ using Buster. I receive the following error:
```
Installing NextCloudPi...
ERROR: distro not supported:
Debian Buster with Armbian Linux \l
```

nachoparker, 2019-07-21:
it works for me. Anybody else seeing this
- use the NextCloudPi issue tracker
- share the contents of `/etc/issue`

jimgregory, 2019-07-21:
I tried again with a fresh Armbian install and had the same problem.  I've posted my problem and the contents of `/etc/issue` on your NCP issue tracker.