armbian/config #68: Strange GUI

jotoeri, 2019-09-06:
Hi all,
i just updated my BananaPi to Armbian Buster (Armbian 5.91).
The configuration utility there seems to be broken in its 'Graphics' (see picture). The submenus look alike, even that special submodules as the Timezone-Module still look good.
Unfortunately i don't know directly, where this could come from, so i can't fix it by myself. :/
Greets,
Jonas

[image]

jotoeri, 2019-09-06:
Ahhh....! I just forgot: By moving the cursor it gets even worse... ;)
[image]

igorpecovnik, 2019-09-06:
> The configuration utility there seems to be broken

Known bug. Not ours - some core component or terminal handling. But if solution is found, we will adopt a workaround.

https://packages.debian.org/buster/dialog
https://packages.debian.org/buster/libncursesw6

jotoeri, 2019-09-06:
Ok, nice. Thank you. ;)

closed 2019-09-06 by jotoeri