armbian/config #76: Discussion: drop OMV installation from debian buster

chwe17, 2019-12-19:
OMV no longer provides dedicated arm images based on armbian:
https://sourceforge.net/projects/openmediavault/files/OMV%205.x%20for%20Single%20Board%20Computers/

with version 5 (based on buster) they recommend (https://forum.openmediavault.org/index.php/Thread/28789-Installing-OMV5-on-Raspberry-PI-s-Armbian-Supported-SBC-s/ and in their documentation: https://github.com/OpenMediaVault-Plugin-Developers/docs/blob/master/Adden-A-Installing_OMV5_on_Armbian.pdf) using their own script (https://github.com/OpenMediaVault-Plugin-Developers/installScript/raw/master/install) instead of the IMO more convenient way over armbian-config. From a (really) quick overview I didn't spot a difference between their script and what amrbian-config does when 'converting' armbian into a OMV but I might be wrong here cause I didn't compare it fully. 

Since Thomas stepped back from maintaining OMV on Arm boards the direct communication between both projects seems to be less tight. And (in the future) needed changes to our installation routine on armbian-config may not find its way into our repository. To avoid confusion (was it installed over armbian-config or https://github.com/OpenMediaVault-Plugin-Developers/installScript/raw/master/install) I think it would make sense to drop OMV 5 from armbian-config or the OMV project adjusts their documentation and contributes to the maintenance of this part, to ensure armbian-config 'does the right thing' for a proper OMV  experience. 
As an OMV user since years I would like to see it back in armbian-config with an adjusted documentation but from a maintenance perspective I don't see a reason to keep it when the main project (OMV) specifically for armbian doesn't  recommend to use it.

//cc
@ThomasKaiser 
@flmaxey (uploaded the document which doesn't recommend armbian-config)
@ryecoaaron (author of the current recommended installation routine)

ryecoaaron, 2019-12-19:
I didn't write the script to replace armbian-config's OMV option.  It was written to have something that worked on all platforms (armbian, RPi's raspbian, x86, etc).  I probably should've submitted an issue and/or pull request for the things that armbian-config is doing wrong with OMV 5.x.  But my time is limited and I didn't want to maintain the code in two places.

chwe17, 2019-12-20:
> things that armbian-config is doing wrong with OMV 5.x.

can you shortly elaborate what armbian-config currently is doing wrong? As for all of us, the time is always limited so with your help we might fix this without much work on both sides. I would happily fix it and I'm sure it would be nice for your users as well when they just can call armbian-config to convert armbian into a OMV installation. I wont replace Thomas as a OMV maintainer on arm, but I can contribute to keep armbian-config up to date with your needs for OMV.

flmaxey, 2019-12-20:
> From a (really) quick overview I didn't spot a difference between their script and what amrbian-config does when 'converting' armbian into a OMV but I might be wrong here cause I didn't compare it fully.

Perhaps a closer comparison of the script and armbian-config might reveal the differences.

ryecoaaron, 2019-12-21:
- The big one is it should use "omv-salt deploy run" to configure services since omv-mkconf doesn't exist on OMV 5.x.
- It is changing afp settings but the netatalk plugin doesn't exist on OMV 5.x.
- chrony is the name of the ntp service now instead of ntp
- There is no need to enable the usul-proposed repo since there is nothing in it and can be enabled from the web interface.

ryecoaaron, 2019-12-22:
Another issue that I am working on with my script is removing network-manager since it causes problems with docker (which a large majority of OMV users use).  Even Docker recommends disabling network-manager - https://success.docker.com/article/should-you-use-networkmanager

g-provost, 2020-01-23:
We exchanged couple of emails with @ryecoaaron in order to discuss how to spin this topic into something else than "_Open Discussion: drop OMV installation from debian buster_" and make it constructive. I won't hide, my goal to improve the collaboration is because OMV is an important app for boards like Helios4 and Helios64.

First we need to understand the OMV priority in term of board support, and accept that their biggest SBC user group is using RPi boards. Therefore if they are going to maintain an install script it is normal for them to main a script that support most use cases but before all their biggest user groups.

So instead of maintaining a piece of script on our side that will fall out of scrutiny form the OMV team, why not changing the approach and making omv_install func call their install script (https://github.com/OpenMediaVault-Plugin-Developers/installScript/raw/master/install). I mean as said by @chwe17 the delta between is pretty minimal.

In omv_install func we only keep board related tweak, and whatever system tweak we still feel are necessary and not provided by install script from OMV.

I think if we go that @ryecoaaron is keen to get more involve on our side and help us maintaining omv_install function part. Better he confirms himself here.

As for [InstallOpenMediaVault()](https://github.com/armbian/build/blob/master/config/templates/customize-image.sh.template#L37) in customize-image.sh, I think we all know that OMV never installed flawlessy within chrooted environment. So yeah maybe this piece will require some major rework or be set as deprecated. I still think both side can benefit of being able to provide Armbian images with pre-installed OMV... but ok maybe not the highest priority. First we need to fix armbian-config.

So if on our side we are keen to leverage on OMV install script, I will start work with @ryecoaaron to make the changes.

Please use thumps up / down to vote.

ryecoaaron, 2020-01-23:
@g-provost I am fine with this plan.

g-provost, 2020-01-28:
So I guess since no reaction = no thumbs down, therefore we should move ahead with suggested rework of omv_install and leverage on external OMV install script.

g-provost, 2020-03-18:
Currently even install of OMV4 on Stretch is broken, because it depends on ntp package and Armbian has traded ntp for crony. No big deal, but the install script doesn't support yet the situation. There is other issue too, we are looking at it.

igorpecovnik, 2020-03-18:
> Anyway instead of fixing it, let's fix the whole thing once for all. And do as suggest that omv_install func call their install script. We will do a PR soon.

O.K.

g-provost, 2020-03-23:
Let's close this discussion and let's hope this is not an issue that resurface again in the future. There still some work to be done after PR #86 
1/ What to do with InstallOpenMediaVault() function in customize-image.sh, should we just remove it or mark as deprecated ?! 
2/ the tweak applied to Helios4 are not persistent across OMV update. It's not critical right now but definitively need to be address soon.

closed 2020-03-23 by g-provost

cmoulliard, 2021-01-19:
OMV dont appear at all from the softy list using rockpro64 where Armbian Focal has been installed. Should I then install manually OMV5 (as I suppose OMV dont support this debian distro) ?

g-provost, 2021-01-19:
OMV is only supported on Debian.

- Armbian Stretch = Debian
- Armbian Focal = Ubuntu

cmoulliard, 2021-01-19:
> OMV is only supported on Debian.

So, which NAS solution does armbian recommend then ?

g-provost, 2021-01-19:
If you want to install OMV, then use Armbian Stretch images.

igorpecovnik, 2021-01-19:
> So, which NAS solution does armbian recommend then ?

My personal recommendation (and how I use NAS) is manual setup without any additional garbage:
https://www.digitalocean.com/community/tutorials/how-to-set-up-a-samba-share-for-a-small-organization-on-ubuntu-16-04

There are many other NAS solutions out there. All will work on Armbian but you have to use their support. If they say its made for Debian, get Armbian based on Debian. If they say its working under Docker, enable Docker. We don't set those rules, 3rd party software developers do.