armbian/config #80: armian-config sid package forcefully pulls in iptables

Avamander, 2020-01-20:
It would be really nice if iptables were an optional port of armbian-config so that people could also use nftables if they wish to do so.

EvilOlaf, 2020-01-20:
It seems like (for the moment at least) there are some functions that rely on iptables. But yes, you are right. iptables will be deprecated somewhen in the future and replaced with nftables.
I tried my first steps with nftables in another case a few days ago and failed for some reason.

igorpecovnik, 2020-01-20:
Agree. But I do lack time already to maintain basics of this tool. Any help will help me to do more about.

Avamander, 2020-01-20:
I'd just really love if it didn't hard-depend on iptables, I'm pretty okay with some specific functionality failing if I can keep the rest.

igorpecovnik, 2020-01-20:
Dependencies cleanup https://armbian.atlassian.net/browse/AR-50 was planned for 20.02 but it was moved to 20.05 but even for that I am not sure I will manage to do anything about.

igorpecovnik, 2020-12-03:
https://github.com/armbian/build/commit/7427b0b0c47cd5bfa7ac8946c81a4fee1751d78b

closed 2020-12-03 by igorpecovnik