armbian/config #88: Source install fails

Averell7, 2020-04-08:
armbian-config used on a Asus Tinkerboard. 
Menu software / source_install. 
When the work is finished, in the /usr/src/linux-source-5.4.28-rochchip directory, there is a single file, .config, nothing else. 
software / header_install work properly
The linux-source...._all.deb file is correctly downloaded, I could get it from /var/cache/apt/archives.

The script indicates that it is uncompressing, configuring, and there is no error message. May be a simple error in the command which (I guess) would delete the tar files, and which deletes everything ?

igorpecovnik, 2020-04-08:
Test resoults shows sorces has been succesfully installed. Are you using updated images from the download section or your own builds?

https://dl.armbian.com/_test-reports/2020-03-31_01.03.57.html

> May be a simple error in the command which (I guess) would delete the tar files, and which deletes everything ?

https://github.com/armbian/config/blob/master/debian-config-jobs#L187-L229

igorpecovnik, 2020-04-08:
Tested now manually - it works as expected. Closing.

Feel free to open if it doesn't work on a clean build.

closed 2020-04-08 by igorpecovnik

Averell7, 2020-04-08:
I just installed armbian-config today, after an apt update. The Tinker board linux is downloaded from your website : Armbian Buster with Linux 5.4.14-rockchip
I verified :
The source is correctly downloaded. 
The two .xz files are correctly installed in /usr/src
I have not yet been able to see if at some stage the package is really extracted. 
In the end, only the .config file remains.

igorpecovnik, 2020-04-08:
Did you reboot after updating?

Averell7, 2020-04-08:
Yes, I rebooted. But I found the trick (not tested yet but quite sure this is the reason). pv was not present on my system. So please, add it to armbian-config dependencies.

Averell7, 2020-04-08:
OK, it works now.