armbian/config #92: Error when installing hassio from armbian-config

estalldecker, 2020-04-17:
Hi 

I have installed recently a fresh version of Armbian in a cubitruck I have. Today (2020-04-17) the installation of hassio failed but without any message from armbian-config. After looking for a while at the cause I detected that 'softy' ('debian-software') installer is trying to get an installer file from github but is getting a 404. 
The file that the installer fails to fetch is this one: 
`https://raw.githubusercontent.com/home-assistant/hassio-installer/master/hassio_install.sh`
in line 785 of debian-software script:
`curl -sL "https://raw.githubusercontent.com/home-assistant/hassio-installer/master/hassio_install.sh" | \`

I fixed it by pointing to this install script:
`https://raw.githubusercontent.com/home-assistant/supervised-installer/master/installer.sh`

Please let me know if you want a pull request to merge this. 
Thanks for the wonderful work and excellent distro!

Regards
Emmanuel

igorpecovnik, 2020-04-17:
> Please let me know if you want a pull request to merge this.

PR is always thankful - it saves us some time! 

> Thanks for the wonderful work and excellent distro!

Double thanks :)

closed 2020-04-18 by estalldecker

LazaroFilm, 2022-09-08:
I have the same issue on my Orange Pi Zero 2. selecting Hassio from the software suite in Armbian-config does nothing.

EvilOlaf, 2022-09-09:
https://github.com/armbian/config/pull/167