armbian/config #95: Board performance configuration caps CPU frequency below maximum on Odroid HC2

adatum, 2020-05-02:
On an Odroid HC2 with armbian buster + openmediavault 5 + armbian-config, the default board configuration upon installation is for the XU4. After selecting the HC1/HC2 configuration with armbian-config, the maximum frequency for CPUs 4-7, which have a nominal 2000 MHz frequency, is limited to 1700 MHz. Is this known or intentional? 

Benchmarks:
HC1/HC2 configuration: http://ix.io/2h7g
XU4 configuration:  http://ix.io/2k57

igorpecovnik, 2020-05-02:
This is properly functional only on stock kernel, not on 5.4.y ... here it should be disabled until someone ports special configurations for HC1/HC2. Going back to XU4 is the only strategy here. There are some minor stability issues with this 5.4.y kernel ... https://forum.armbian.com/forum/11-other-supported-boards/ but it seems solved. At least in the dev lab.

adatum, 2020-05-02:
Thanks, I've switched back to the default XU4 board config for now. At least now I know.

Perhaps this issue can be updated if/when the configuration is ported for the HC1/HC2.

igorpecovnik, 2020-05-02:
> Perhaps this issue can be updated if/when the configuration is ported for the HC1/HC2.

Yes. I hope someone will prepare DT based on old kernel. There is no much difference, but it is some work, focus to adjust namings and test. 

Leaving open.

adatum, 2020-05-02:
I would offer to help, but I don't know much on this topic. I did not even know about 5.4.y vs stock kernel.

I can help with providing information or relatively safe testing.

igorpecovnik, 2020-05-02:
> I would offer to help

Ofc. Even I would need to invest some time to research. I only know where approximately too look for ...

There are many ways to help:
https://github.com/armbian/build#contribute
:)

closed 2020-07-07 by igorpecovnik

adatum, 2020-07-07:
Why wontfix?

igorpecovnik, 2020-07-07:
> Why wontfix?

Because this and next year I have no time to do anything about. You are welcome to hire someone and submit a fix to public software - this way: https://docs.armbian.com/Process_Contribute/

adatum, 2020-07-07:
> Because this and next year I have no time to do anything about. 

Why not just leave the issue open?