armbian/config #98: armbian-config depends on resolvconf

belegdol, 2020-05-20:
I have recently upgraded to openmediavault-5.4. It appears to use systemd-resolved by default. This appears not to be compatible with armbian-config as armbian-config depends on resolvconf which I believe would conflict with systemd-resolved.

ryecoaaron, 2020-05-20:
In my opinion, armbian-config should not be used once OMV is installed.  They will cause conflicts with each other (as you found).

igorpecovnik, 2020-05-20:
Its planned to clean this up a bit [AR-50](https://armbian.atlassian.net/browse/AR-50) to prevent some troubles, but yes, when using OMV or similar tool that also fiddles (heavily) around the system its better not to use this tool.

belegdol, 2020-05-20:
I found armbian-config useful in the past to e.g. enable HC1 config or switch the kernel branches. For some reason (probably me not knowing better) when I originally installed omv 4.x my odroid was set to standard xu4 mode and was using -next kernel.

closed 2020-12-01 by igorpecovnik