armbian/config #99: can't install anything

moeinrahimi, 2020-05-24:
I'm trying to install docker from softy section, when I select docker softy page reloads and nothing happens. I'm using a tinkerboard.
my uname output:
```
Linux tinkerboard 5.4.28-rockchip #20.02.7 SMP PREEMPT Sat Mar 28 14:49:13 CET 2020 armv7l GNU/Linux
```

igorpecovnik, 2020-07-07:
Docker is supported on all kernels but install is highly dependent from OS layer. To be able to resolve a bug it is important to know which userspace you were using - Debian Stretch, Buster, Ubuntu Bionic, Focal, ... ?

jeankhawand, 2020-09-28:
I am facing same Issue I am running Focal 20.04 
`Linux orangepizero 5.8.5-sunxi #20.08.1 SMP Wed Sep 2 08:03:45 CEST 2020 armv7l armv7l armv7l GNU/Linux
`

igorpecovnik, 2020-12-03:
https://github.com/armbian/config/commit/22e8ac6f39bc19183e4eb106041c5d26458dafc2

closed 2020-12-03 by igorpecovnik