armbian/config #108: SimpNAS Installed

johnnyq, 2020-07-21:
Hi I'm the developer for SimpNAS and opensource NAS similar to openmediavault https://simpnas.com is there any way we could get our installer integrated into armbian config if so whats the best way to go about doing so. Below is a link to our installer and works very well with armbian.

https://github.com/johnnyq/simpnas/blob/master/install.sh

EvilOlaf, 2020-07-22:
The integration itself should be pretty easy. A simple example for an existing one whould be here: https://github.com/armbian/config/blob/e2bba42b02fbde3b3a4c3178d74d00c14d1c193a/debian-software#L647-L654
Grab it, install it, done.

From there it gets a bit more complicated for two reasons: 
First it installs docker and samba which armbian-config also offers and could conflict with each other if users selects those for installation later on. This needs to be catched.

Also the installation ideally needs to be tested on a wide varity of boards and that takes time which is a rare and valuable resource....

lanefu, 2021-04-15:
@johnnyq any interest in pursuing @EvilOlaf 's suggestions?