armbian/config #109: Cannot set static IP

JPluess, 2020-07-24:
I have successfully connected to my wife, but setting a static IP fails. 
[image]

I tried to save like so:
[image]

Why does it say adapter lo above?  Isn’t it wlan0? Like so:
[image]

thanks, ;)

igorpecovnik, 2020-07-24:
> I have successfully connected

Thanks for reporting an issue. We will look into it when possible.

> Why does it say adapter lo above? Isn’t it wlan0? Like so:

Perhaps you used some other tool to connect or its simply a bug that doesn't count that? Support terms in general https://github.com/armbian/build#support

EvilOlaf, 2020-07-24:
Tried the same with OrangePi Zero. It automatically selects eth0 for me when selecting "static". So maybe there is an issue with interface selection?

JPluess, 2020-07-24:
I re-burned this image:  Armbian_20.05.2_Orangepilite2_buster_current_5.4.43.img with Rufus

Right after booting and setting up my account, I only used armbian-config tool to make my settings

1. changed time zone, Zurich
2. change locales de_CH.UTF-8 and set as default
3. changed keyboard to CH
4. changed mirror to Denmark
5. connected to my WIFI

  **6a. Failed to change IP settings.
6b. After reboot issue persists.**

7. I did an update / upgrade command:
[image]

8. after a reboot I tried again to set IP, with no success
[image]

igorpecovnik, 2020-07-24:
> Failed to change IP settings.

Yes, we register it as a bug.

igorpecovnik, 2020-07-24:
armbian-config is a wrapper for other low level apps .... try using:

      nmtui-edit

for setting fixed IP.

lrhorer, 2020-10-10:
Confirmed this does not work on the latest version of Buster.  /etc/network/interfaces is ignored.  The only way I was able to set a static IP on either WiFi or Ethernet was to remove network-manager and its libraries, install dhcpcd5 and edit /etc/ /etc/dhcpcd.conf.  For wireless, I also had to create /etc/wpa_supplicant/wpa_supplicant.conf.

igorpecovnik, 2024-09-23:
New mechanism has been implemented into https://github.com/armbian/configng

closed 2024-09-23 by igorpecovnik