armbian/config #111: I2C Clock Speed

nogleo, 2020-08-11:
Noob here!

I need to change the Clock Speed from the I2C bus on my OrangePi running Armbian. I've seen in the datasheet the board is able to use FastMode (400 kHz). How Am I suposed to do that?

Thanks a lot in advance

martinayotte, 2020-08-11:
See the following post on the forum (search would have reveal it) :+1: 
https://forum.armbian.com/topic/3030-h3-i2c-speed/?do=findComment&comment=21132

nogleo, 2020-08-11:
I saw the post previously and by my understanding it would take building the OS to make this change, correct? Forgive me if I'm wrong.

I just wanted to know if there is, like in the RPi for example, a way to change it after built like the config. txt or a change in parameter in a file

igorpecovnik, 2020-08-12:
This problem is not related to armbian-config.

closed 2020-08-12 by igorpecovnik

martinayotte, 2020-08-12:
As the post explained, with Mainline kernel, you only need to edit DT, no need to rebuild kernel.

nogleo, 2020-08-12:
Sorry but I don't even know what DT stands for... total noob here

EvilOlaf, 2020-08-12:
> 
> 
> Sorry but I don't even know what DT stands for... total noob here

`Device Tree` and please refer to the Armbian forums to further discussion. As mentioned earlier this issue tracker is for the armbian-config tool only.

igorpecovnik, 2020-08-12:
>  total noob here

https://linuxjourney.com/
https://github.com/armbian/build#support

spyderdyne, 2024-03-22:
> See the following post on the forum (search would have reveal it) 👍 https://forum.armbian.com/topic/3030-h3-i2c-speed/?do=findComment&comment=21132

Search revealed this issue thread. This is why everyone should be on board with just providing answers if they are going to post en lieu of snarky criticisms or posting links for further research. FWIW