armbian/config #116: Which type of Wifi?

salimbasar, 2020-10-19:
I have an orange pi 4 . it has AP6256, IEEE 802.11 a/b/g/n/ac, 

I have created a hotspot by armbian-config. ı have got directly as 5G but in our system there are devices that not connecting to 5g.

How we can change this to 2,4 and 5g together. or only 2,4 ghz mode.

here is my hostapd.conf 
`#
# armbian hostapd configuration example
#
# nl80211 mode
#

ssid=bTalkerNET
interface=wlan0
hw_mode=g
channel=40
#bridge=br0
driver=nl80211

logger_syslog=0
logger_syslog_level=0
wmm_enabled=1
wpa=2
preamble=1

wpa_psk=66eb31d2b48d19ba216f2e50c6831ee11be98e2fa3a8075e30b866f4a5ccda27
wpa_passphrase=12345678
wpa_key_mgmt=WPA-PSK
wpa_pairwise=TKIP
rsn_pairwise=CCMP
auth_algs=1
macaddr_acl=0

## IEEE 802.11n
ieee80211n=1
ht_capab=[DSSS_CK-40][HT20+]
country_code=US
ieee80211d=1
## IEEE 802.11n

## IEEE 802.11a
hw_mode=a
## IEEE 802.11a

### IEEE 802.11ac
#ieee80211ac=1
#vht_capab=
#vht_oper_chwidth=1
#vht_oper_centr_freq_seg0_idx=42
### IEEE 802.11ac

# controlling enabled
ctrl_interface=/var/run/hostapd
ctrl_interface_group=0`

igorpecovnik, 2020-12-10:
> here is my hostapd.conf

Technical support = https://forum.armbian.com

> Is it possible to add select type possibility type duringhot spot creation?

This feature is not implemented.

igorpecovnik, 2024-09-23:
New mechanism has been implemented into https://github.com/armbian/configng

closed 2024-09-23 by igorpecovnik