armbian/config #118: Home Assistant doesn't start on OrangePiPC

Frans1960, 2020-11-01:
After installation trough ArmBian-config of Home Assistant, home assistant doesn't start. Docket is installed and it looks like Home Assistant is installed, but it's not running in a Docker Container.

EvilOlaf, 2020-11-01:
What `armbian-config` basically does is call their install script and execute it. So maybe HASS made some changes upstream to break the installation on your particular board.
https://github.com/armbian/config/blob/cf3874a8d5c3b38b6858106aeff93d9bdbb296bf/debian-software#L757-L791

Frans1960, 2020-11-02:
I've solved it, after a lot of trying I found out that it was a bad UTP cable.
That give a lot of strange problems with installing software.

closed 2020-11-02 by Frans1960