armbian/config #130: Wrong Version Number 21.11.4 for armbian-config package?

wodev, 2020-12-20:
It looks like you published the latest armbian-config package with the wrong version number **21.11.4** instead of **20.11.4**.

```
# apt-get upgrade -s
Reading package lists... Done
Building dependency tree
Reading state information... Done
Calculating upgrade... Done
The following packages will be upgraded:
  armbian-config
1 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Inst armbian-config [20.11.3] (21.11.4 Armbian:buster [all])
Conf armbian-config (21.11.4 Armbian:buster [all])
#
```

igorpecovnik, 2020-12-20:
Sh*, thanks.

igorpecovnik, 2020-12-20:
Removed, correct version is being synched up but we might need to implement a fix somewhere.

wodev, 2020-12-20:
yes, if the package is already installed, they have to downgrade to the correct version otherwise new updates are blocked for the next year :-(

I think it can be solved with following steps:

1.  update the apt package index :
```
apt-get update
```
2.  Downgrade to correct version
```
apt-get install armbian-config=20.11.4
```
3.  Purge the 21.11.4  version from cache ( /var/cache/apt/archives )
```
apt-get autoremove
```

igorpecovnik, 2020-12-20:
> new updates are blocked for the next year :-(

Yeah. We could deal with this via BSP package hack. But this is not a critical problem anymore. Just a problem.