armbian/config #132: Add cockpit to softy

lanefu, 2020-12-23:
Another great option for quick home server stuff.    Maybe we can install cockpit with some optimized configs.     Has a subset of OMV's capabilities and can be installed on focal and buster

https://cockpit-project.org/running

lanefu, 2021-04-12:
some what amusing.. OMV provides this as a plugin now

igorpecovnik, 2021-04-12:
OMV provides "everything" :) which means we still can provide things separate. And OMV is buster only.

lanefu, 2021-04-12:
Ah yes great point.

ryecoaaron, 2021-04-12:
Not that you care but omv-extras added cockpit as a button to install to replace virtualbox (since it is not in buster repos).  omv jokes aside, the version of cockpit in the regular buster repo is ancient, looks very different, and is buggy.  The buster-backports version is current.  The focal version is fairly old too but should still be usable.

lanefu, 2021-04-12:
@ryecoaaron Dude... we totally care!

good info..   I find it amusing that Cockpit does all this cool stuff and doesn't have official support for samba config.    I saw there's a plugin a while back, but not sure if it's maintained anymore.  You know anything about that?

ryecoaaron, 2021-04-12:
Assuming you are talking about https://github.com/enira/cockpit-smb-plugin, I think I tried it out a long time ago on Fedora but haven't looked at it recently.  Doesn't look like it is maintained anymore.

djismgaming, 2023-04-02:
This one is currently used for Samba: https://github.com/45Drives/cockpit-file-sharing

Needs this one to maintain Samba passwords: https://github.com/45drives/cockpit-identities

This one is great to have a directory navigation webUI directly in cockpit that can be used to fix permissions in the whole filesystem, download/upload files and more: https://github.com/45Drives/cockpit-navigator