armbian/config #137: Softy installs not doing anything

SpontaneousDuck, 2021-02-07:
So I just installed Armbian Buster 21.02.1 on my ODroid HC1 board and using `armbian-config`. I am unable to install anything with softy. I can see the list of options and select software to install but when I select it, it says "Loading software info..." and then fails back to the software list silently without installing anything. Is there somewhere I can go to see the logs of what this is doing and what is failing? 

`uname -a` output: `Linux odroidxu4 4.14.212-odroidxu4 #12 SMP PREEMPT Wed Feb 3 21:31:20 CET 2021 armv7l GNU/Linux`

Thanks for the help!

igorpecovnik, 2021-02-07:
Do you select software with a spacebar?

Otherwise join forums - there its support going on. Here just to file an issue that can be identified.

We are seeking for ideas and people to refactor this tool. We would like to have it, its handy, but if has to be redesigned and maintained properly. https://forum.armbian.com/topic/16933-armbian-config-rfc-ideas/

SpontaneousDuck, 2021-02-07:
Ah, there it is! Spacebar needs to be sued to select the software you want before selecting install. Thanks for the help! I would also like to help out with the work on the tool. I'll head over to that thread.

closed 2021-02-07 by SpontaneousDuck