armbian/config #142: functions-network force nameserver 8.8.8.8 into both static and dhcp configurations, which is problemtic

crosser, 2021-04-03:
There were a (now closed) ticket [build 1096](https://github.com/armbian/build/issues/1096) about this problem, and several forum questions. However, the problem still exists in focal, and problematic code is in the master branch.

The problem manifests as the address 8.8.8.8 being the first `nameserver` in the DNS resolution order. This is wrong on several levels:

1. Ideologically, not everyone is comfortable with outsourcing name resolution to an external commercial enterprise, operating in a particular jurisdiction,
2. Security-wise, the user can run validating resolver in their network, but armbian devices will use different resolver, that is not validating,
3. Practically, the user's resolver may support local TLDs, such as ".lan" for example. Armbian devices are not able to resolve hosts in such domain.

Problematic code resides in [/lib/armbian-config/functions-network.sh](https://github.com/armbian/config/blob/master/debian-config-functions-network)

lanefu, 2021-04-12:
Hi @crosser 

Yes it's silly to set DNS to 8.8.8.8.   The scope of this issue specifically seems to when setting static IPs via `armbian-config` rather than something native such as `nmtui`

Functionality should be added to let the user specify DNS when setting a static IP.

The file that would need to be edited in this repo is https://github.com/armbian/config/blob/master/debian-config-functions-network

Given the workarounds (use native tools for configuring network)  It's a low priority from an Armbian Development perspective,.  but we'd be more than happy to review and accept any PR's for improvements.

lanefu, 2021-05-05:
resolved via #143

closed 2021-05-05 by lanefu