armbian/config #146: Keyboard-setting can't be changed

rabelux, 2021-06-04:
In personal settings hitting the keybard-option shows "loading keyboard submodule" and then returns to the previous screen.
There is nothing that I can do here. It says my keyboard layout is US but when typing it's DE as set in locales (y and z aren't swapped)
[image]

If you need more infos let me know. I'm new to armbian and started setting up my NanopiM4 headless server today.
OS: Armbian 21.05.2 stable
Armbian-Config: 21.05.3

EvilOlaf, 2021-06-04:
Can you execute `dpkg-reconfigure keyboard-configuration` and see what happens?

rabelux, 2021-06-04:
Well nothing happens. It's busy for a second then returns.

igorpecovnik, 2021-06-04:
Perhaps this helps https://benohead.com/blog/2012/04/27/debian-change-the-keyboard-layout-from-the-console/

rabelux, 2021-06-04:
Just to be clear:
The layout works as expected and is DE, it's just that armbian-config displays US.
As I've just booted a fresh install of armbian which I dowloaded [here](https://www.armbian.com/nanopi-m4/#kernels-archive-all) I thought I'll let you know.
Not a major bug, just a little confusing.

Oh and now when I entered "personal settings" `Locales` language preview was empty. Then I entered it, went up and down, pressed <kbd>esc</kbd> and now I have `C.UTF-8` as displayed language. I did't press anything else, just up und down, no enter.
Something weird is going on here...
[image]

W-Fonseca, 2022-01-17:
have you tried **_sudo nano /etc/default/keyboard_** and change settings? 
Link: https://docs.armbian.com/User-Guide_Fine-Tuning/

EricaLinaG, 2023-08-13:
Im experiencing this as well.
It doesnt matter what I Change it to or how I Change it, it doesn't actually change.
Mine is in an unusable state.

EricaLinaG, 2023-08-13:
I have overcome this to some degree by using setxkbmap from the command line.
However I have to do that every time I login. 
I get multiple warnings about more than one variant defined. Over riding xserver.

I have tried to find where it actually is defined with no luck. It is not defined in the standard places. /etc/X11 is where it should be. 

Basically my computer is unusable until I override the keyboard settings which I can't find and armbian-config has no effect over. Although that is what set it this way in the first place.

Tearran, 2024-02-23:
My understanding X11 keyboard settings are handled at the X11 user level, and `setxkbmap` is the right tool for changing these settings from the command line.

To make the change permanent, you can add the `setxkbmap` command to one of the user's X startup scripts. For example, if you want to set the keyboard layout to US English, you can add the following line to `~/.xinitrc` or `~/.xsession`:

`setxkbmap -layout us`

Tearran, 2024-03-01:
line [#854 ](https://github.com/armbian/config/blob/master/debian-config-jobs) 
```
dpkg-reconfigure keyboard-configuration 
``` 
Does not seem to do anything.

When looking for available maps
```
localectl list-keymaps
Failed to read list of keymaps: No such file or directory
```
```
localectl status
System Locale: LANG=en_US.UTF-8
    VC Keymap: (unset)
   X11 Layout: us
    X11 Model: pc105
```

```
localectl set-keymap us
Failed to set keymap: Keymap us is not installed.
```

sudo nano /etc/default/keyboard
Edit XKBLAYOUT=
```

# KEYBOARD CONFIGURATION FILE

# Consult the keyboard(5) manual page.

XKBMODEL="pc105"
XKBLAYOUT="us"
XKBVARIANT=""
XKBOPTIONS=""

BACKSPACE="guess"
```

| Country           | XKBLAYOUT Code | XKBVARIANT Code | Confirmed |
|-------------------|----------------|-----------------|:---------------:|
| US                | us             |                 | :heavy_check_mark: |
| UK                | gb             |                 | |
| Germany           | de             |                 | |
| Canada (English)  | ca             |                 | |
| Canada (French)   | ca             | fr  |
| China             | cn             |                 | |

EricaLinaG, 2024-07-16:
I need to do this again, so I'll have to remember how I got it working. I'll give it a try on my new vim4.

Tearran, 2024-08-04:
Used to reconfigure the keyboard layout settings on a Debian-based system.
```bash 
sudo dpkg-reconfigure keyboard-configuration
```
Used to apply the console font and keyboard layout settings. 
 It is typically run after 'dpkg-reconfigure keyboard-configuration' to ensure the changes take effect.
```bash
setupcon
```
Updates the initial RAM filesystem (initramfs). Necessary to ensure that any changes to the system configuration, such as keyboard layout, are included in the initramfs, which is used during the boot process.
```bash
sudo update-initramfs -u
```
files that may also be relative
/etc/default/locale
/etc/locale.gen