armbian/config #147: armbian-config fails on Orange Pi Zero (Buster 5.10.43)

moracabanas, 2021-07-01:
After fresh install i get this error trying to open `armbian-config`

```log
root@orangepizero:~# armbian-config
Warning: Configuration cannot work properly without a working internet connection.      Press CTRL C to stop or any key to ignore and continue.Command line is not complete. Try option "help"
Error opening terminal: unknown.
'unknown': I need something more specific.
Error opening terminal: unknown.
```

EvilOlaf, 2021-07-01:
Did you connect it to internet properly?

moracabanas, 2021-07-01:
Googling for setup my wifi suggest to use `armbian-config` to setup wifi on some systems, documentation was not clear. 
Lastly I found `nmtui` the proper way to setup wifi and now `armbian-config` is working properly.
IMHO `armbian-config` is confusing:
- It gets annunced on every boot
- So you try to use `armbian-config`, it won't work and you will think why I am trying to config wifi with this tool and it actually ask for internet 🙃
- The first time some users boot they use USB serial TTY to setup wifi.
- `nmtuy` is pretty straight forward, why is not advertised on boot, or on `armbian-config` fail?

Thanks you for taking your time reviewing this.

igorpecovnik, 2021-07-01:
> IMHO armbian-config is confusing

Agree. We have nice plans to transform it into non confusing professional tool https://forum.armbian.com/topic/16933-armbian-config-rfc-ideas/ just the problem is that we are totally busy and it is estimated that 1 - 2 full timers are needed for 6 - 12 months to get it to functional level. While we are already totally busy maintaining this OS ... just raising the money and just finding people in the times when corporations with endless access to money have troubles getting them ... is needed to start moving into that direction ;) 

> The first time some users boot they use USB serial TTY to setup wifi (me).

This is rarely used method - normal tri pin serial connection is still a gold standard in this world.

Wifi chip on Orangepi zero is troublesome, check forums https://forum.armbian.com/forum

moracabanas, 2021-07-01:
Thanks you for your time and your elaborated response 😀

closed 2021-07-01 by moracabanas