armbian/config #152: Unable to install Docker from

Pacman85, 2021-09-07:
Hi 
I am new to Armbian and I am impressed over it compared to Tinker OS for tinker Board that I was using before. 
I really liked the armbian-config and 3 party installer softy. 

Only problem I have found so far is that I can not install docker via softy? 
Are there other steps that I need to do to be able to install that program?

Thanks

igorpecovnik, 2021-09-07:
> I am new to Armbian 

Read slowly:
https://docs.armbian.com/User-Guide_FAQ/
https://github.com/armbian/build#support

> compared to Tinker OS for tinker Board

Investments into Armbian exceeds TinkerOS for magnitude. Armbian is non-profit OS and it serves you, not the purposes of sales (TinkerOS).

> Only problem I have found so far is that I can not install docker via softy?

Please check this wizard on how to properly report bugs:
https://www.armbian.com/bugs

Armbian comes in many variant and Docker install differs from that. If we don't know everything, nobody will even look into the bug. "No logs, no bug".

> I really liked the armbian-config and 3 party installer softy.

We also like it, but it got very expensive to maintain it or to provide any end users support. In the FAQ you can see why. Enjoy ;)

Pacman85, 2021-09-09:
Thank you for the information. I will post on the right channels from here on out.

closed 2024-09-23 by igorpecovnik