armbian/config #157: Support OMV 6 Install on Debian 11 Bullseye

Lyteral, 2021-11-21:
Hi all,

with OMV 6 now being released, it would be great if the armbian-config tool would enable the install of OMV 6 on supported OS platforms (Debian 11 Bullseye)

Currently, armbian-config checks explicitly for Debian 10 "Buster" being installed and then launches the OMV-Install script (which as far as I can tell currently only supports OMV5)

To support the OMV 6 install:
- it would be required to update the checks in the Armbian-Config tool (either by letting the user decide between OMV 5 and 6 or by selecting the correct version according to the OS)
- it would be required to adapt the external OMV-install script used by armbian-config to support the OMV6 install path 

Best regards
Daniel

ryecoaaron, 2021-11-21:
> then launches the OMV-Install script (which as far as I can tell currently only supports OMV5)

The install script does support OMV 6.x  The debian-software script needs to allow bullseye though.

hendriebosch, 2022-01-05:
Hello,

For me this still doesn't work. When armbian-config is used to install omv it responds with the old dialog: OpenMediaVault 5 can only be installed on Debian Buster.

There has been an apt update and an apt dist-upgrade and it says it's up to date.

What to do?

Kind regards,
Hendrie

Lyteral, 2022-01-05:
Hi Hendrie,

that change is not in a released version of armbian-config yet.

It should work if you git clone and then use the „Development version“ as shown in the Git readme.md

Lyteral, 2022-01-07:
> Hello,
> 
> For me this still doesn't work. When armbian-config is used to install omv it responds with the old dialog: `OpenMediaVault 5 can only be installed on Debian Buster.`
> 
> There has been an `apt update` and an `apt dist-upgrade` and it says it's up to date.
> 
> What to do?
> 
> Kind regards, Hendrie

Did my proposed solution work for you? If not, feel free to reopen the Issue.
In some time the support for OMV6 will also be available in the "release" version of armbian-config in the APT repositories

closed 2022-01-07 by Lyteral