armbian/config #179: DTB package not reinstalled on kernel switch

seth-hunter, 2022-12-03:
Commit 9c793ad4c863af4b919ca4cf7f0afe5e777eb03f placed dtb installation behind an erroneous apt-search.  apt does not support searching by version, therefore this check always fails and dtb package gets removed but not reinstalled. This is causing devices running 2022.11 that switch kernel versions to fail to boot.

seth-hunter, 2022-12-03:
Tagging @igorpecovnik - this may be considered urgent as it is bricking 2022.11 systems that attempt to switch kernels via armbian-config.

igorpecovnik, 2022-12-03:
ok, will be pushed out asap, takes up to 48h to sync with all servers ... (also we are working to get this time down to <1h by smart repository enablement)

closed 2022-12-03 by igorpecovnik

igorpecovnik, 2022-12-03:
synching ... not in index yet.
https://fi.mirror.armbian.de/apt/pool/main/a/armbian-config/

PR #180 (2022-12-03): fix: dtb not installed during kernel switch
Fix #179. The prior if block already checks if the specified version of dtb package exists, so move the variable append to that block instead of performing a second check that doesn't work.