armbian/config #181: gpg: can't create '/etc/apt/keyrings/docker.gpg': No such file or directory at fresh armbian install

yswtrue, 2022-12-10:
When I install docker by armbian-config get these error:
```
gpg: can't create '/etc/apt/keyrings/docker.gpg': No such file or directory
gpg: no valid OpenPGP data found.
gpg: dearmoring failed: No such file or directory
(23) Failed writing body
```
And I create folder `/etc/apt/keyrings` and re-run again it works again.

iav, 2022-12-10:
What OS based your armbian image? Ubuntu/Debian, and version number?

yswtrue, 2022-12-10:
Armbian: 22.​11.​1 ​bullseye current
board: rockpi 4c+
I use this image: Armbian_​22.​11.​1_​Rockpi-​4cplus_​bullseye_​current_​5.​15.​80.​img.​xz

iav, 2022-12-10:
ok, thank you.
Do you need a workaround or have docker working already?

yswtrue, 2022-12-10:
Docker is working now.

PR #182 (2022-12-11): Fix #181: create folder for keyring if not exists
Fix #181: create folder for keyring if not exists
commit 86233039 (2022-12-11): Fix #181: create folder for keyring if not exists (#182)
closed 2022-12-11 by igorpecovnik