armbian/config #190: System→Freeze(Disable Armbian kernel upgrades) not work for new kernel naming

iav, 2023-04-01:
System→Freeze(Disable Armbian kernel upgrades) not work for new kernel naming scheme.
 

Despite of just chosen "Freeze" function `# apt list --upgradable` shows 

```
linux-dtb-edge-meson64/jammy 23.05.0-trunk--6.2.9-Se128-De8b2-P018a-C6f95Hfe66-Ve377-Be9aa arm64 [upgradable from: 23.05.0-trunk--6.2.9-Se128-D2a58-P8aa5-C1c9dHfe66-Bee7e]
linux-headers-edge-meson64/jammy 23.05.0-trunk--6.2.9-Se128-De8b2-P018a-C6f95Hfe66-Ve377-Be9aa arm64 [upgradable from: 23.05.0-trunk--6.2.9-Se128-D2a58-P8aa5-C1c9dHfe66-Bee7e]
linux-image-edge-meson64/jammy 23.05.0-trunk--6.2.9-Se128-De8b2-P018a-C6f95Hfe66-Ve377-Be9aa arm64 [upgradable from: 23.05.0-trunk--6.2.9-Se128-D2a58-P8aa5-C1c9dHfe66-Bee7e]
```

rpardini, 2023-04-03:
Hey,
- how does armbian-config freeze stuff? If it's `apt-mark hold xxx`...
- ... that doesn't prevent possible upgrades from being listed in `apt list --upgradable`` ...
- ... but does prevent and actual `apt upgrade` from upgrading them.

Not saying we don't have some kind of problem with the version, but...

iav, 2023-04-07:
```
sudo apt upgrade
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Calculating upgrade... Done
The following packages have been kept back:
  qemu qemu-user-static
The following packages will be upgraded:
  bind9-dnsutils bind9-host bind9-libs containerd.io distro-info-data dnsutils
  docker-buildx-plugin docker-ce docker-ce-cli gh imagemagick
  imagemagick-6-common imagemagick-6.q16 libldb2 libmagickcore-6.q16-6
  libmagickwand-6.q16-6 libwbclient0 **linux-dtb-edge-meson64
  linux-headers-edge-meson64 linux-image-edge-meson64** linux-tools-common
  python3-ldb python3-samba samba samba-common samba-common-bin samba-libs
  samba-vfs-modules
28 upgraded, 0 newly installed, 0 to remove and 2 not upgraded.
Need to get 152 MB of archives.
After this operation, 1,169 kB of additional disk space will be used.
Do you want to continue? [Y/n]
```
I add `**` to mark  **linux-dtb-edge-meson64
  linux-headers-edge-meson64 linux-image-edge-meson64** inside terminal output

rpardini, 2023-04-07:
Ok so this is about armbian-config and how it actually freezes the packages. Do you know where the source for that feature lies?

iav, 2023-04-08:
There https://github.com/armbian/config/blob/ce33cd38097a247158ca40746187c44732415699/debian-config-jobs#L958-L977

iav, 2023-04-08:
Seems it should work now.
But it doesn't.
And possible I found a reason: I use an `edge` kernel now. but `${BRANCH}` in check code looks for `current`. 
Highly likely™ in past I just deinstall old kernel packages and install self-builded edge. 

Then now question is "what should be a right".

iav, 2023-04-08:
Should this code still look for branch at some config? or try to look for real?

rpardini, 2023-04-08:
Yeah, I dunno. I'm under strict medical orders to not look into armbian-config source, so I can only speculate that it looks at the info produced by the bsp-cli package in /etc/armbian-release and similar. so if you installed an edge kernel with apt it might be... confused.

iav, 2023-04-08:
okay.
Thank you for co-operations.
seems nothing should be fixed there.
Sorry for disturb.

closed 2023-04-08 by iav

rpardini, 2023-04-08:
🖖  You don't disturb at all. We've also `BSPFREEZE=yes` in build system so this actually helps