armbian/config #194: DTC not working on RK3588 5.10.xxx

NicoD-SBC, 2023-06-13:
When going to System -> DTC on the RK3588 then you only get an empty nano window.
Using the Mekotronics R58 Mini with Armbian Jammy Gnome 5.10.160-rockchip-rk3588
This is what messages I get.
```
SoC manufacturer: rockchip
board name: rk3588-hugsun-minipc-linux
FATAL ERROR: Couldn't open "": No such file or directory
stat: cannot statx '/tmp/tmp.8A9CyDcpFo/current.dts': No such file or directory
stat: cannot statx '/tmp/tmp.8A9CyDcpFo/current.dts': No such file or directory
Do you want to edit the device tree again? (y/n)
```

igorpecovnik, 2023-06-13:
Can you try again after installing dtc;
sudo apt install device-tree-compiler

NicoD-SBC, 2023-06-13:
> Can you try again after installing dtc; sudo apt install device-tree-compiler

`nicod@mekotronics-r58-minipc:~$ sudo apt install device-tree-compiler 
[sudo] password for nicod: 
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
device-tree-compiler is already the newest version (1.6.1-1).`

igorpecovnik, 2023-06-14:
Not a trivial solution then :(

rpardini, 2023-06-14:
Confirmed. The "DTC" editing code in armbian-config was introduced here https://github.com/armbian/config/pull/123 and lives mostly here https://github.com/armbian/config/blame/ed8fe64aae4bbf3e6cf775ff26b6ebb6c0c8d129/debian-config-jobs#L1519-L1647

@igorpecovnik it already does install `dtc` if not installed.

This code is trying to read the compatible from `/proc/device-tree/compatible`, find/compute the `soc_manufacturer` from that, then find the board based on ?, and find the DTB file that is actually used via glob matches. This is very error prone...

As we can see from @NicoD-SBC's logs, with the rk3588 legacy kernel (and probably many, many others) the compatible doesn't match, errors are ignored (as always in armbian-config), everything fails but an editor opens with an empty file.

Sincerely we should drop all this, read `/boot/armbianEnv.txt` for the `fdtfile=` which will get us directly to the .dtb that is in use. (for extlinux, parse `/boot/extlinux/extlinux.conf` for the first `fdt` stanza amounts to the same).

rpardini, 2023-06-14:
Alternatively: we can get the running DT blob (whatever it is, wherever it comes from) via

```shell
dtc -I fs /sys/firmware/devicetree/base
```

jeanrhum, 2023-07-27:
@rpardini Sorry for answering late. I was very busy the last months...
I agree that using the extconf fdt field info is better. However, if I remember well some (old supported) boards was not using this file when I wrote this code (but maybe I am wrong).
If all officially supported boards now use this file to specify the dtb, I think that the modification you suggest should be used. 

I can try to update the code in this way, but I haven't any rk3588 board to test it. I'll only test on my boards based on other socs (Allwinner h3 h6, Amlogic s805x a311d, Rockchip rk3328 rk3399).