armbian/config #196: dts editor hardcoded to nano

zerog2k, 2023-06-24:
dts editor seems to be hardcoded to `nano`

https://github.com/armbian/config/blob/master/debian-config-jobs#L1605

should respect `/usr/bin/editor` and/or $EDITOR

iav, 2023-06-25:
`sensible-editor`, part of debian package `sensible-utils` - sensible editing

SYNOPSIS
**sensible-editor** [OPTIONS...]

DESCRIPTION
       sensible-editor  makes  sensible  decisions  on which editor to call.  Programs in Debian can use this
       script as their default editor.  sensible-editor try to do in the following order:

1      if VISUAL environment variable exists, execute VISUAL [OPTIONS]
2      if EDITOR environment variable exists, execute EDITOR [OPTIONS]
3      source the contents of file ~/.selected_editor and, if SELECTED_EDITOR environment variable ex‐
              ists execute SELECTED_EDITOR [OPTIONS]
4      run editor [OPTIONS] command
5      finally run nano-tiny [OPTIONS] command

viraniac, 2024-01-03:
This should be working now.

closed 2024-01-03 by viraniac

lanefu, 2024-01-03:
Resolved via https://github.com/armbian/config/pull/200