armbian/config #216: [NanoPi R5S | Rockchip64] Other > Branch Switch incomplete transition

HeyMeco, 2024-03-30:
When I switched my NanoPi R5S from current to edge branch that only installed `linux-dtb-edge-rockchip64` and `linux-image-edge-rockchip64` and `linux-u-boot-nanopi-r5s-edge` from edge but still did not remove the old ones except for u-boot.

Also `armbian-bsp-cli-nanopi-r5s-current` is on my board and not the -edge variant which I also couldn't manually install via apt.

amazingfate, 2024-03-30:
No need to remove, you can keep multi kernel packages in your system.

HeyMeco, 2024-03-30:
@amazingfate Yes I manually removed the old ones since the emmc isn’t that large on NanoPi R5S but armbian-bsp-cli confused me since there is also an -edge variant which didn’t replace the -current one. Also I couldn’t really find any info on what it does / if it’s needed.