armbian/config #226: Odroid M1, Network Adapter Performance, in jammy

rvalle, 2024-09-11:
Hi!

I am looking into OdroidM1 network performance with kernel 6.6.47-current-rockchip64 ( 24.8.1 )

For some reason network performance is x10 faster (upstream) in bookworm and noble, when compared to jammy.
Downstream performance is fine, delivering the same performance in jammy and the other distributions.

Since all of them are using the same kernel, I assume that there must be some configuration that jammy is missing, or?

Any idea about what could be going on?

EvilOlaf, 2024-09-11:
This issue tracker is for bugs of the build framework, issues when _building_ to say.
Issues _running_ Armbian should be posted in forums: https://forum.armbian.com
https://www.armbian.com/bugs/

closed 2024-09-11 by EvilOlaf

rvalle, 2024-09-11:
Thanks @EvilOlaf , so many places to push issues to.... was difficult to decide.

igorpecovnik, 2024-09-11:
Jammy, Noble, Bookworm ... are not different in this. Pay attention to kernel version. Its a known regression, also found on official Hardkernel images. No solution yet.